lathoub/Arduino-BLE-MIDI #1: BLE MIDI implementation help

alf45tar, 2018-11-01:
I am using your awesome AppleMIDI library for my Pedalino project.
Nice to see you started a BLE MIDI library. I already implemented BLE MIDI in Pedalino, not as a library of course, but I think you can re-use a lot of source code to speed up your job.
I will be happy to move to you library when it will be released.

Cheers
alf45tar

lathoub, 2018-11-02:
Hi alf45tar

Thank you for offering help, much appreciated - all help is welcome. Thanks for the kind words.
I'm very impressed by your [Pedalino](https://github.com/alf45tar/Pedalino) work, and see indeed that you have already implemented the MIDI BLE stack. Nice!

Do I understand it correctly that you are looking to offload your MIDI BLE to this library so that you can focus on the Pedalino, without having to maintain the MIDI BLE code yourself? If so, I'd be more than happy to welcome your code.

I'm currently looking into aligning the MIDI interface code from the 47effects, AppleMIDI and this library. (Even going so far as have an additional library with the stubs for MIDI - I see you have also created all the codes, `MidiType` (in `MidiTimeCode.h`), ... that can be shared between the projects. Not sure if 47effects would go along. Just an idea at this stage) 

I look forward to your comments and help
Lathoub

alf45tar, 2018-11-02:
Yes, I am looking for a BLE MIDI library and you have the required expertize to create the reference library for BLE MIDI like you did for RTP-MIDI. 
I started implementing BLE MIDI because I didn't find anything ready on the shelf.
If you can take care to it I can focus on Pedalino.

The API aligning job is very appreciated too.

Do you want me to extract the BLE code from Pedalino source code or is it clear to you what to keep and what to delete?

Thanks
alf45tar

lathoub, 2018-11-02:
OK, understood - sounds like fun.
I will have a first stab at extracting your code from Pedalino and ask you for feedback.

lathoub, 2018-11-03:
I should also align better with the [APIStyleGuide](https://www.arduino.cc/en/Reference/APIStyleGuide) and [LibraryTutorial](https://www.arduino.cc/en/Hacking/LibraryTutorial) (eg on function naming: capitalise the first character? Use of `underscore` in function and file names)

alf45tar, 2018-11-03:
Good job!

lathoub, 2018-11-03:
Thanks @alf45tar your code was nicely segmented, I only had to cut and paste.
I just introduced an abstract class for MIDI input and output - BLE specialised it, I will also add it to the AppleMIDI library (and possibly later also the FortySevenEffects/arduino_midi_library

Can you check the concept of the abstract midi interface lib and give feedback?

lathoub, 2018-11-03:
generic MIDI interface in its own Git: https://github.com/lathoub/MIDI

lathoub, 2018-11-11:
Skeleton finished, for most messages (not sysex).  @alf45tar , can you verify if the current version fits in Pedalino?

alf45tar, 2018-11-12:
It works like a charm. I am using it for my new project PedalinoMini (https://github.com/alf45tar/PedalinoMini) but I will use it for Pedalino too.

I suggest to rename sendMidi method because it doesn't send anything but it triggers callbacks on receipt of MIDI messages.

One more thing. If you want to complete your portfolio of MIDI libraries you should create one for ipMIDI protocol. It should be not too hard because the code is in my Pedalino and you already own the abstract MIDI framework.

Very good job.

closed 2019-07-11 by alf45tar