lathoub/Arduino-BLE-MIDI #3: BLE MIDI not sending any MIDI signal out

bondrezz, 2018-12-11:
Hi, 
This library is awesome! Before, I couldn't manage to discover the BLE device on IOS/OSX. But with your library, It work like a charm! However, When I tried to send a midi note, I didn't receive any Note on Ios/Osx. But I can read the note from Ios/Osx just fine.

I'm using this code to send a midi note
bm.sendNoteOn(60, 127, 0) ;
delay(1000);
bm.sendNoteOn(60, 127, 0) ;
delay(1000);

Is there anyway for me to send a note/ CC value?

dacodekid, 2019-02-18:
@bondrezz See if [this pull](https://github.com/lathoub/Arduino-BLE-MIDI/pull/6/commits/79c7d4ec22f0a1ea37c609d5304e585cc89a565b) request helps

lathoub, 2019-02-23:
Fix by @dacodekid accepted. Are you happy with the fix @bondrezz ?

bondrezz, 2019-02-24:
@lathoub Yes, I'm very happy. And I've noticed that channel have to be 1-16 not 0. Thank youuu
@dacodekid Thank you very much.

closed 2020-10-03 by lathoub