lathoub/Arduino-BLE-MIDI #8: Send/receive SysEx Messages

R0Wi, 2019-08-04:
Thank you for this nice library, i'm using it for a while now and it works great!

The only thing i'm missing is to receive sysex message via the BLE interface. So i have seen that there are stubs to be implemented in the code for sending and receiving. Is there any chance to get this implemented? Could i help somehow?

Thanks for considering.

lathoub, 2019-08-05:
Hi @R0Wi ,
I currently do not have a BLE device to play with, but feel free to add your code!
Bart

spotman, 2020-09-29:
@lathoub first of all, huge thanks for this library! I see you are working on the `feat/2.0.0` branch right now. Can we elaborate on SysEx messages on that branch? They are very useful for transmitting non-music payload (patches / firmware update / etc) and I would be happy to help :slightly_smiling_face: 

Cheers,
Denis

lathoub, 2020-10-03:
Hi @spotman I just cleaned `feat/2.0.0` and moved it into master. I'll bring over the SysEx example from USB-MIDI (that sends SysEx from the Arduino), receive will be in another example. Happy to accept all help with testing and coding

R0Wi, 2020-10-03:
Sound great i'm also looking forward to this and i can help testing and improving 👍

lathoub, 2020-10-03:
**Sensing** SysEx complete (incl over multiple packets) - can you test @R0Wi  and @spotman? (see notes in my last commit for the edge cases)

spotman, 2020-10-04:
@lathoub sorry for being late, I'll check everything ASAP, tomorrow probably. Thank you for your time and attention to this issue! Have a nice day!

R0Wi, 2021-07-29:
Hi @lathoub and thanks for adding this feature. Sorry for the delay but now i finished my "BLE MIDI proxy" project (https://github.com/R0Wi/Arduino-ESP32-Midi-BLE-Proxy) which noch uses the latest version `2.1.0` of your library. The sysex messages work just fine like you also described in your examples so now my project is really complete :sunglasses: 

I think this issue can be closed now, thanks again !

closed 2021-07-29 by R0Wi