lathoub/Arduino-BLE-MIDI #9: Crash conflict with ESP Async Webserver

psgcooldog, 2019-09-18:
I'm working on a piece of hardware/software to control a TC-Helicon VoiceLive 3 Extreme.  It has become a fairly large project, and I need BLE-MIDI, and a good web server.  ESP Async Webserver seems to be the popular one.

I had BLE-MIDI, and a ton of other stuff working, then started adding the web server.  The web server would crash the ESP32, fail to respond to a request, or stop responding after one request.

By a process of elimination, I've established that if the BLE-MIDI is not started, then the web server is just fine.  Can you point me to any particular place to start debugging?

I am working with PlatformIO.

psgcooldog, 2019-09-23:
I now get error messages from the wifi initialization, where it's attempting to allocate 16 buffers but only getting 13, and also "E (17803) SPIFFS: cache buffer could not be malloced"

Something in BLE-MIDI is screwing up the heap, I think.

lathoub, 2020-10-11:
This lib has been totally redesigned - and this issue was not follow-up upon, so will close it. Should the issue come up again, happy to look into it again.

closed 2020-10-11 by lathoub

cloverstreet, 2021-08-02:
I had similar experience with BLE-MIDI and ESPNOW.  Switched to another library, can't recall who made it, but the libraries are these...
#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_ESP32_NimBLE.h>
Solved my crashing problems