lathoub/Arduino-BLE-MIDI #14: Buffer effect

alf45tar, 2020-04-26:
I recently migrated my PedalinoMini project to the latest Arduino MIDI Library v5 using AppleMIDI, ipMIDI and BLE as transport layers.

AppleMIDI is working well as usual. ipMIDI works very well considered it is pretty new.

BLE is not working very well right now under stress situation. It works but there is a delay on receiving/sending MIDI messages.
The first messages are transmitted in real time but after some messages in a short period messages are delayed.
I am using the MIDi Wrench to sending notes swiping from one end to the other of the keyboard.  Pedalino is continuing receiving notes after some seconds I finished to swipe on the keyboard. It happens only with feat/2.0.0 branch.

PedalinoMini is working at light speed with master branch of Arduino-BLE-MIDI.

Thanks
alf45tar

lathoub, 2020-04-26:
Thanks @alf45tar for the feedback. feat/2.0.0 uses the ESP32 xQueueSend and xQueueReceive for cross core communication - when sending a lot of messages, it seems to create delay. I will need to look into it

lathoub, 2020-04-26:
What if you would increase `MaxBufferSize` in `BLE-MIDI_Settings.h` from 64 to 256

alf45tar, 2020-04-27:
No improvement at all.

lathoub, 2020-04-29:
can you try to set `Use1ByteParsing` in `Settings` to false in the MIDI library and see if that makes a difference?

alf45tar, 2020-04-29:
How can I modify setting? I don't see any BLEMIDI_CREATE_CUSTOM_INSTANCE?

Thanks

lathoub, 2020-04-29:
The setting sits in the dependant MIDI library (so not in this library) - in the `midi_Settings.h` file

lathoub, 2020-05-01:
I have updated feat/2.0.0 so that `Use1ByteParsing` is now set to false in the setting.
When `Use1ByteParsing` is `true` `MIDI.read` parse only one byte at a time, that is not optimal for transport layers that reads more bytes at a time (BLE is a packet transport protocol)

alf45tar, 2020-05-01:
Sorry but it is the same. Use1ByteParsing makes no difference.
Are xQueueSend and xQueueReceive mandatory? Can we return to master branch BLE implementation?

lathoub, 2020-05-01:
an ESP32 has 2 cores - the BLE runs on core 2 and the MIDI code runs on core 1 - and synchronisation is needed.

How much data to you pump through? xQueueSend and xQueueReceive should be able to send/receive 10.000 bytes in 30ms.

The master-branch should be unchanged - xQueueSend and xQueueReceive are only in the feat/2.0.0 branch.

alf45tar, 2020-05-01:
Can we use the same implementation used for master branch without xQueue for feat/2.0.0?
I am using master branch on ESP32 without any issue.

lathoub, 2020-05-01:
master branch is **not** based on the Arduino MIDI Library v5
You might not notice, but the eg NoteOn events come from another core than core that run the loop() (on ESP32, BLE runs on Core 0 - Arduino setup/loop runs on Core 1 - (I have it the other way around above))

alf45tar, 2020-07-12:
Issue disappeared after heavy updates to PedalinoMidi. It looks like it is not an issue of this library but an interaction with other libraries.

closed 2020-07-12 by alf45tar