lathoub/Arduino-BLE-MIDI #15: Error compiling for board DOIT ESP32 DevKit V1

felixed, 2020-06-18:
Hi, 

I'm stuck with this error when trying the example Esp32_NoteOnOffEverySec:

> Arduino: 1.8.12 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"
> 
> In file included from /Users/felixed/Documents/Arduino/libraries/Arduino-BLE-MIDI-master/src/BleMidi.h:14:0,
>                  from /Users/felixed/Documents/Arduino/libraries/Arduino-BLE-MIDI-master/examples/Esp32_NoteOnOffEverySec/Esp32_NoteOnOffEverySec.ino:1:
> /Users/felixed/Documents/Arduino/libraries/Arduino-BLE-MIDI-master/src/Ble_esp32.h:9:31: fatal error: common/midiCommon.h: No such file or directory
> compilation terminated.
> exit status 1
> Error compiling for board DOIT ESP32 DEVKIT V1.

What can I do? I can successfully compile and run other programs on this board, so I'm assuming my toolchain works. As far I can tell common/midiCommon.h is a dependency and so far I haven't been able to find how to include it.

Thanks in advance.

Jardu, 2020-10-02:
Hi lathoub and felixed,
first of all thank you for making this library @lathoub. I really would like to try some things with it, but I'm also stocked by the same error message felixed mentions above. What is this "common/midiCommon.h" file. I can't locate it. Could this be general thing? 
Thank you for any comment on the subject.
Best regards.
Jardu

lathoub, 2020-10-02:
try out branch feat/2.0.0 that links to the MID library
https://github.com/lathoub/Arduino-BLE-MIDI/tree/feat/2.0.0

I'm about to merge, so your input is welcome

Jardu, 2020-10-02:
Hi lathoub,
thank you for your quick response. I recently installed both libraries (USB-MIDI & MIDI Library) and the directory path should be ok.  As far as I understand, there are still header files missing: The same way felixed mentioned, namely the "common/midiCommon.h" for «Esp32_NoteOnOffEverySec.ino» and compiling «MidiBle.ino» I got: ...Documents/Arduino/libraries/Arduino-BLE-MIDI-feat-2.0.0/src/hardware/BLEMIDI_ESP32_NimBLE.h:4:26: fatal error: NimBLEDevice.h: No such file or directory, compilation terminated.
I can't finde those header files locally. Do you have a suggstion? I'm I on the wrong track.
Thank you in advance. 
Best regards.
Jardu

P.D.: I'm using the Arduino app version 1.8.10 & 1.8.13

lathoub, 2020-10-02:
BLE-MIDI_Config.h is still in development - i just fixed the #defines - you should be OK for the ESP32

Jardu, 2020-10-02:
Hi lathoub,
thank you so far for your effort. I noticed some conflicts with my installed MIDI libraries on my computer, so I deactivated them and also installed the NimBLE-Arduino library separately. Finally I could compile and upload the MidiBle.ino! Yeah :-) The other sketches «Esp32_NoteOnOffEverySec.ino» ´n «Basic_IO.ino»  are still not compiling. But thank you so much, it definitely helps being able to use a Midi-Library!!!! 
Best regards and all the best.
Jardu

lathoub, 2020-10-03:
Cleanup and all committed to the master branch - only basic example for the moment.
When you are happy, pls close this issue

Jardu, 2020-10-04:
Thank you lathoub. As far as I understand I cannot close the issue  by myself since I didn't open it?!
Best regards
Jan

closed 2020-10-05 by lathoub