lathoub/Arduino-BLE-MIDI #17: Replace Bluedroid with NimBLE

alf45tar, 2020-09-15:
What do you think to add support for NimBLE-Arduino?
It looks like not too much work but a lot of fun.

https://github.com/h2zero/NimBLE-Arduino
https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/Migration_guide.md

Cheers
alf45tar

alf45tar, 2020-09-15:
I already did the changes in less than 10 minutes. PedalinoMini worked at the first try. The result are as promised.
Let me know if you want me to submit a PR or do you prefer to do yourself?

alf45tar, 2020-09-15:
I did 3 changes in hardware/ESP32.h file:

1. Replaced the existing #include with just one #include <NimBLEDevice.h>
2. Find 'BLECharacteristic::PROPERTY_' and replaced with 'NIMBLE_PROPERTY::' at line 221, 222, 223 and 224
3. Removed the line 227 _characteristic->addDescriptor(new BLE2902());

Of course you can do a better job adding a conditional compiling to support both: Bluedroid and NimBLE.
For the moment I don't think I will return to Bluedroid.

Enjoy
alf45tar

lathoub, 2020-09-16:
cool, thanks @alf45tar  i've uploaded a first version to feat/2.0.0 (based on your recommendation) - can you test it?
(it worked on my Mac)

alf45tar, 2020-09-16:
Tested on PedalinoMini and it works like a charm.

closed 2020-09-16 by alf45tar