lathoub/Arduino-BLE-MIDI #18: No connection from M5 Atom (ESP32-PICO-D4)

scottmetoyer, 2020-09-17:
I'm trying to get the example sketches running on an M5 Atom development kit, which is based on an ESP32-PICO-D4 .

Everything builds but I'm not able to connect to the Atom from other Bluetooth devices - it doesn't show up in the list of available Bluetooth devices.

Does this library support the ESP32-PICO-D4, or am I doing something wrong? Thanks!

lathoub, 2020-10-02:
No clue - i don't have one :-).
 (assuming you use feat/2.0.0 branch)

I have developed it using Huzzah32 Feature and wESP32

lathoub, 2020-10-03:
Feat/2.0.0 rolled to master, pls check again

scottmetoyer, 2020-10-06:
Still no luck, but I realized I can't connect via Bluetooth at all from the device (it doesn't appear in my Mac's list of nearby Bluetooth devices) so there is likely something else going on. Thanks for checking!

lathoub, 2020-10-11:
I just with 2 EPS32 Boards (Feature and wESp32) and BLE-Midi connects just fine. Can you try again with the latest version (and basic example)

lathoub, 2021-09-23:
Any update @scottmetoyer ; are you able to connect with the current code in the master branch? (are you using Windows?)

closed 2021-10-07 by scottmetoyer