lathoub/Arduino-BLE-MIDI #25: Trouble on macOS / Logic Pro

xelarep, 2020-12-20:
Hi, created a simple Mackie HUI Recording Light on an ESP32 DevKit. While programming/debugging I used PocketMIDI and an IAC connection to monitor in/outgoing MIDI messages (I forward the MIDI data from Logic by the Thru function to the ESP32...) . Everything is fine and works as expected in this configuration.
However if I want to directly connect in Logic to the ESP32 by BT the ESP does not react at all. What can I do?

lathoub, 2020-12-20:
> the ESP does not react at all. 
Are you to connect to the ESP32?

xelarep, 2020-12-20:
Yes, connection ist Ok, Logic can select the ESP. However there’s no data transfer (I guess no input at ESP as there’s no reaction on the same data from a different sender...)

lathoub, 2020-12-20:
What sketch are you running?

xelarep, 2020-12-20:
[https://github.com/xelarep/RecordingLightHUI](https://github.com/xelarep/RecordingLightHUI)- just a simple sketch - current work state...

lathoub, 2020-12-20:
See if you can get the simple demo to work (MidiBle) and try another tool (eg I'm using MIDI Wrench on my iPhone)

xelarep, 2020-12-20:
As said: this sketching working if I put PocketMIDI via IAC between logic and the ESP32. Also standalone if I send manually the needed commands by PocketMIDI data entry... this sketch is based on the MIdiBle sample...

xelarep, 2020-12-21:
And BTW: my sketch is also working with an iOS device + MIDI Wrench...
Only a direct Logic Pro connection is not working. I have to put data out from Logic to the IAC driver. And the in PocketMID connect the IAC as input port by the thru function to the ESP out port
[image]

xelarep, 2020-12-21:
One more observation: if I close PocketMIDI in my running environment (i.e. closing IAC in and ESP32 BT out ports, still holding an active BLE connection) and try to change the HUI output port in logic, it takes some seconds and Logic reports one less available  MIDI in and a out port. In MIDI-Studio, the list of BLE devices gets greyed out. I have to power cycle my ESP32 for the next reconnect attempt. 
If I just close and reopen PocketMIDI (i.e. I break/reopen the connection from Logic thru IAC, Pocket MIDI to my device) everything is fine and Logic transmits updated data, and my device still works without problems.

xelarep, 2020-12-21:
Hm, one more weird thing: for testing purposes I grabbed one of my Leonardos and replaced the MIDI BLE by MIDI USB (also removed the OnConn../OnDisco...). Same behaviour over USB MIDI?!

xelarep, 2020-12-21:
Last update for today: after stumbling through some boxes of electronic parts and sorting out a broken DIN cable I was able to run the same sketch (i.e. only modification for MIDI on HW serial...) on standard DIN-MIDI over my UR22mkII MIDI interface. This time I was able to connect direct from Logic / MIDI ports of the UR22... 
So something with connection management on macOS seems to be part of my problem...

closed 2021-08-06 by lathoub