lathoub/Arduino-BLE-MIDI #26: Devise not advertising

BruceWayne-cpu, 2021-02-09:
I'm trying to use this library with a Wemos D1 mini 32, but when I try with NimBle example it compiles and uploads without errors but it does not show up in my BLE scanner.
However when I try with the Basic example with "#include <hardware/BLEMIDI_ESP32.h>" it throws:

[image]

lathoub, 2021-02-09:
I don't have any Wemos D1 hardware, so can't compare.  The Wemos D1 mini is ESP8266 based, not? From the log: **device not support** error, that would explain why it does not compile using the Esp32 header.

BruceWayne-cpu, 2021-02-09:
The Wemos D1 mini 32 is esp32 based. The strange thing is that I've already used the esp 32 BLE libraries successfully in this same board. I mean:  <BLEDevice.h>, <BLEUtils.h>, <BLEServer.h>, <BLE2902.h> they work fine, but when using <hardware/BLEMIDI_ESP32.h> it throws that error...

BruceWayne-cpu, 2021-02-09:
Whats interesting is that when I look into the libdeps folder in the project that uses  <BLEDevice.h>, <BLEUtils.h>, <BLEServer.h>, <BLE2902.h>, there are no dependencies related to BLE, maybe they are core drivers? 
But when I look into the libdeps folder in the BLE-MIDI project there is an ArduinoBLE library.

BruceWayne-cpu, 2021-02-09:
Why is it ArduinoBLE a dependency?

BruceWayne-cpu, 2021-02-10:
Ok, so I removed the ArduinoBLE library that was automatically installed by Platformio/Arduino IDE (tried with both) and it works fine now. Both the NimBle version and the normal one. Maybe consider removing that lib dep? thank you for your good work!

lathoub, 2021-02-10:
if you use NimBLE and the ESP32 BT stack, you indeed do not need ArduinoBLE. But if you use the Arduino Nano 33 BLE, you do need the ArduinoBLE stack. Unfortunately, the IDE does not allow for conditional dependencies.

A solution to avoid the conflict would be to remove the dependencies and have them installed manually - depending on the platform and prefered BLE stack.

BruceWayne-cpu, 2021-02-11:
Yes, i think that could be a solution.

sdstronic, 2021-02-11:
Hi , working on the same configuration, i post here
My ESP32 is ok on Midi with full stack BLE, but very high memory consomption...
I try BLE-MIDI and after manage dependency issue, all seems compiled fine
but i get: E NimBLEAdvertising: "Advertisement data length exceded" on the MIDI.begin();
any idee ?

AlexFundorin, 2021-03-19:
I have the same issue with generic ESP32 board from Aliexpress.
It was working with ArduinoBLE, but I decided to try your library today.
The same errors, after I initialized the library like this: 
```
#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_ESP32.h>
```
Dunno why  is says "Unsupported board" to a "esp32doit-devkit-v1".

lathoub, 2021-12-09:
https://github.com/lathoub/Arduino-BLE-MIDI/issues/31

lathoub, 2022-01-17:
stale

closed 2022-01-17 by lathoub