lathoub/Arduino-BLE-MIDI #27: setHandleNoteOn called when it shouldn't

aphleps, 2021-03-08:
I'm using the Arduino-BLE-MIDI library in my project, latest version, it's almost identical to the MidiBle.ino example. On my computer I have a 1 bar MIDI loop going in Ableton Live. Everything works for some time, but every now and then I get two weird Note-On events.

`
HUB :: Note-On          channel: 2      note: 61        velocity: 100
HUB :: Note-Off         channel: 2      note: 61        velocity: 64
HUB :: Note-On          channel: 2      note: 129       velocity: 71 <- weird
HUB :: Note-On          channel: 2      note: 64        velocity: 145 <- weird
HUB :: Note-On          channel: 2      note: 61        velocity: 100
HUB :: Note-Off         channel: 2      note: 61        velocity: 64
HUB :: Note-On          channel: 2      note: 68        velocity: 100
HUB :: Note-Off         channel: 2      note: 68        velocity: 64
`

The notes in my clip are 61, 57, 68 and 71 and have a velocity of 100. You can see that after a note-on message a note-off if received. In my clip there is no note 129 with velocity 71, and there is no note 64 with velocity 145 (obviously).

RobertoHE, 2021-07-29:
I have seen the same bug too.

lathoub, 2021-07-29:
DAW's tend to send a lot of MIDI messages, could you add a `Serial.println` at L108 in BLEMIDI_Transport.h (and enable logging in setup) and send the dump

https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L103-L110

lathoub, 2021-07-29:
> I have seen the same bug too.

do you have the exact same setup @RobertoHE ?

RobertoHE, 2021-07-29:
Hi @lathoub 

I apologize for the lenght of this message.

I have seen the bug in 2 differents ways: the first one with ServerNimBLE, and the second one with my Client proposal (pull request).

In the server Mode (BleNimble), I used the following setup:

Server (ESP32 ServerNimBLE):
-board = esp32dev
-framework = VSCode/PlatformIO (arduino)
-Code: Basic example with noteOn, noteOff and AffterTouch callback handlers. Loop() only contains Midi.read(). 

Client:
-DAWCubase 
-Win10
-BLE (using loopMidi in the Middle)
-MidiController: Roland AX-EDGE by USB (DAW in the middle) or Casio Pravia PX-560

I generated "Midi Chords" (some notes simultaneously) and some afterTouch automatizations in channel 5 using DAW while playing MidiController generating notes, pitch-bends and aftertouchs messages in channels 1, 2 and 3. The server was in OMNI_CHANNEL mode (receiving 1 ,2 ,3 and 5 channels).
I saw sometimes that same behavour that @aphleps had explained before. I received some notes over 127 pitch or over 127 velocity. I didn't save any logs, sorry. I tried to increment the maxBufferSize, but that did not solve anything (The buffer was never seen overflowed in traces).

In the second setup I used:
Client (ESP - clientNimble):
-board = esp32dev
-framework = VSCode/PlatformIO (arduino)
-Code: Custom code that converts received Midi notes in light patrons using FastLed. Midi.read() is called in a FreeRTOS task, like client example.

Server (AX-EDGE)
-MidiController: AX-EDGE over BLE

This setup is "a little weird" because this client is not in this repo yet, but I hope that you might accept it in the future if it works properly (I am still working on it, but I have seen some lines that I want to change).
The "client version" uses the same midiTransport template and the same Midi decoder than Server Version.

I generated notes, pitch-bends and aftertouchs messages in channels 1, 2 and 3 when I played MidiController (server in this case with BLE notify characteristic) and I could see the NoteOn events using the callback handler. I received in some ocassions weird notesOn events like @aphleps and, at other times, NoteOn callbacks stopped working during severals seconds (between 10s and 90s) but BLE notifications were still received (debug traces were activated and they worked propertly, it still dettected any message generated by MidiController).

I have added some traces in notifyCB(), receive() and read() (and others) functions in Client , Transport and Midi layers and I have seen that BLE notify works fine. NimBLE and notifyCB() still receive incoming data correctly and the data uplinks correctly to the receive() transport layer.

Receive() function adds data to buffer, but I don't know whether it adds correctly F7 (SysEx) type messages. 

I think that the problem may appear when SysEx message comes between other midi messages in the same pakage. I used the callback setHandleMessage() and when this error appears MidiMessage callback object is message.type = F0 (system Exclusive) and message.lenght = 128 (max buffer size in my case). After that, a weird noteOn or other type event is reveived. I think that way of extracting data from the package ( while(true){}, lPtr and rPtr) may origine the issue when F7 appear in the middle of the package.

lathoub, 2021-07-29:
Thanks @RobertoHE 

> I think that the problem may appear when SysEx message comes between other midi messages in the same pakage. I used the callback setHandleMessage() and when this error appears MidiMessage callback [...]

Good catch, this might be the cause of the issue. Let me check over the weekend with the specs. (A message dump would be ideal for me to debug)

RobertoHE, 2021-07-30:
Hi @lathoub 

I have to apologize again for the lenght of the message.

I send you a log of one of the errors. In this case, the error is which cause that parser get crashes and it stops to trigger callbacks (second error of the previous message). 

I used the second setup (Client Mode) that it was explained in the previous comment.
In this case, I only have sent Midi Messages playing the AX-EDGE (only noteOn, noteOff and afterTouch event types in channels 1, 2, and 3 simultaneously, or triplicated). 

In this case, the crash occurred when I was sending a lot of afterTouch events (syntetizer may sends more than 50 events in a second).

I think that BLEMIDI_Transport.receive() has some problems when it receives a second message that arrives while it is processing other one (for example: DAW sends a note and SysEx or other system data in a sort time or, in this case, a lot of affterTouch messages in sort time). Client mode uses a callback that it is trigged automaticly when a notification occurred, and it may cause that two parser may run simultaneously and they may add data at the same time in the RX buffer. I don't sure completly if the bug occurres because the two parser try to add() simultaneously their data in the buffer and they mix the data accidentally, but it may be a possibility.  

After the crash bug, some events continued to arrive and parser sended again a event randomly some times until it started to work fine again. I think that the machine of states of parse() suffers when a bad-created message is added to RX buffer and it tries to process it. Until some "bad midis" pass throught the parser and one of them cleans the machine of states, the parser can't trigger any callback. The extra bad-created messages don't trigger error handler, and it isn't normal and it may be a sympthom.

I have tested a little be other MIDI parser in C++ (https://github.com/TheKikGen/midiXparser) that may do the same that received() in less time. I haven't tested it enoughtly, I will continue to investigate it. But unfortunately, it may present the same inconvenient when it tries to insert data in RX buffer ( add() method).

Alternatively, a flag or semaphore may be added in add() and read() to prevent similtaneous read/write or write/write errors. 

I show you the traces that I added for "debugging".
 
When BLE notification is received (notifiCB()) from AX-EDGE, it prints "-New MSG-" and the incoming data in HEX format. 

I set MIDI callbacks as follow:
```
    MIDI.setHandleError([](int8_t fptr)
                         {
                           Serial.println("\n\nError in parser:" + String(fptr) + "\n\r\n");
                         });

    MIDI.setHandleMessage([](const midi::Message<128U> &msg)  //It is triggered by MIDI.read() when 
      {                                                       //MIDI.parser() process correctly a event
      Serial.println("\n### Parser ###");
      Serial.print("Type: "); Serial.print((msg.type),HEX);
      Serial.println();
      Serial.println("Lenght: " + String(msg.length));
      Serial.println("Channel: " + String(msg.channel));
      });
```

LOGS:
(Bug appear when ###Parser### stops)
```
[...]

-New MSG-
A8 C0 90 3E 3E C1 91 3E 3E C1 92 3E 3E C2 90 40 3E

-New MSG-
A8 C3 91 40 3E C3 92 40 3E

### Parser ###
Type: 90
Lenght: 1
Channel: 1

### Parser ###
Type: 90
Lenght: 1
Channel: 2

### Parser ###
Type: 90
Lenght: 1
Channel: 3

### Parser ###
Type: 90
Lenght: 1
Channel: 1
[... 461 lines cut ...]
-New MSG-
B7 C3 D0 0 C3 D1 0 C4 D2 0

-New MSG-
B7 FE 80 40 57

-New MSG-
B7 FE 81 40 57 FF 82 40 57

-New MSG-
B8 83 80 3E 61 84 81 3E 61

-New MSG-
B8 84 82 3E 61

-New MSG-
B8 C2 80 43 45 C3 81 43 45

-New MSG-
B8 C4 82 43 45 C4 80 47 36 C5 81 47 36 C5 82 47 36

-New MSG-
B8 C6 80 45 4F C6 81 45 4F C7 82 45 4F

```

lathoub, 2021-07-31:
Thanks @RobertoHE , appreciate the long message :-)

> I think that BLEMIDI_Transport.receive() has some problems when it receives a second message that arrives while it is processing other one (for example: DAW sends a note and SysEx or other system [...]

Does this would mean that ESP32's `xQueueReceive` (thread 2) and `xQueueSend` (thread 1) are not fenced and need additional semaphores? (and therefor should work on non-threaded hardware, like a Arduino NANO 33 BLE?)

Or, is it because SysEx messages are intertwined in the normal messages (and would occur on all hardware)?

Where in the above stream are things going wrong?

RobertoHE, 2021-07-31:
Hi again @lathoub .

The last bug is NOT generated by a SysEx. The data streams only had `noteON`, `noteOFF` and `afterTouch` events. I think that the bug is related with 2 or more multiples writer tasks for the same buffer.

I have analyzed `receive()` function and I have some questions, because I am newbie in multi-task RTOS.

In `receive()`, parsed (or accepted) midi package is added to queue byte by byte. For example, a `noteOn` midi is 3 bytes length and `receive()` must call 3 times to `add()` function. In other words, it is needed call 3 differents times to `xQueueSend()` funcion.

`Add()` calls to `xQueueSend()` and copies only 1 byte to `Queue`. When `xQueueSend()` is running, the queue is locked and protected, that it is fine. But when `xQueueSend()` ends, it created some little gaps between a `add()` function and the next one witch `Queue` is unprotected, doesn't?

```
Example of 2 writer threads:

[  ] -> xQueueSend (). queue is locked and protected.
--   -> time between xQueueSend. queue is unprotected.

Thread NotifyCB A: Start |->> Parser ->  Note ON ok   ->  [A0 add] -- [A1 add] -- [A2 add] ->| ends
                                                                               ^    |
                                                                               |    v
Thread NotifyCB B:                  Start |->> Parser ->  Note ON ok   ->  [B0 add] -- [B1 add] -- [B2 add] ->| ends

Expected Buffer : A0 A1 A2 B0 B1 B2
Received Buffer:  A0 A1 B0 A2 B1 B2
```
I am beginner in FreeRTOS, because that, I am not completely sure that `Queue` works at this way or not.

I think that it may be necessary to call `taskENTER_CRITICAL()` or `vTaskSuspendAll()` (or something similar, I am beginner in FreeRTOS) to protect `Queue`.

Midi Messages has a variable length and `Queue` forces to divide messages in bytes to adapt to this variable length. Splitting messages up may mix some packages and it may cause a bug in `Midi.read()`, that it only can process "well-generated" messages.

One inconvenient of `Queue` is that you only can add a fixed size objects to queue, and this doesn't fix well with midi packages. `Stream Buffers` or `Messages Buffers` fix better to variable size packages but they have the inconvenient that not have an implicit protection to lock the buffer for 2 o more writer tasks, and they need `taskENTER_CRITICAL()` or `vTaskSuspendAll()` again.

Ussing `taskENTER_CRITICAL()` may be dangerous with BLE because it disable temporally the interruptions, and it may cause a bug in BLE stack (I suppose it, I am not sure).

I will try to test this hypothesis using my setup (client version).

Regards

lathoub, 2021-08-01:
Thanks @RobertoHE 

> Add() calls to xQueueSend() and copies only 1 byte to Queue. When xQueueSend() is running, the queue is locked and protected, and that's fine. But when xQueueSend() ends, it creates some little gaps between a add() function and the next one in which Queue is unprotected, doesn't it?

True, very good catch!
I create a new [Branch](https://github.com/lathoub/Arduino-BLE-MIDI/tree/ESP32_QueueProt) with a `Semaphore` around the `receive` call (doing the 1 byte at a time adding using `xQueueSend`) and `xQueueReceive`

Admitting, I'm not an ESP32 threading expert, just relying on some old knowledge of thread syncing; **please check**

RobertoHE, 2021-08-01:
Hi @lathoub 

I tried to adapt your code to my client class and it still crashed in the same way.

I captured another log. In this case, parser detected a `middle timeStamp` byte as `SysEx start` wrongly.

I used the same setup as the last test: I only added 3 types of events (`noteOn`, `noteOff` and `afterTouch`). Midi controller coudn't generate a `SysEx` in this setup.

In this case, `receive()` parser miss-interpreted a timestamp that had a valued of `F7` or `F0` and it interpreted this value as a `SysEx start` or `end`. In this case, I think that parser readed a `middle-timestamp` wrongly and timestamp separator had a value over `0x80 (128)`. 

Example MSG: 08 F0 80 34 2B F0 81 34 2B F1 82 34 2B 
Decoder:
	08 F0 -> TimeStamp -> Must be removed
	80 -> Note Off Ch1
	34 -> Note
	2B -> Vel
	`F0 -> TimeStamp -> MUST BE IGNORED -> It is not a SysEx start`
	81 -> Note Off Ch2
	34 -> Note
	2B -> Vel
	F1 -> TimeStamp -> MUST BE IGNORED
	82 -> Note Off Ch3
	34 -> Note
	2B -> Vel

The `receiver()` parser might have interpreted wrongly the timeStamp separator `F0` as `SysEx start`, and it continued processing like `SysEx message` until it received 128 byte (`F7 SysEx` not present and buffer was full).

You can see how parser missed when it returned a type: `F0` (`SysEx`) with a lenght of 128 (max size of `SysEx`). And the follow parsed message did the same.

LOGS:
(remember that some messages may be represented divided by another interruption)
```
-New MSG-
96 F2 D2 3 F4 D0 10 F4 D1 10

-New MSG-
96 F5 D2 10 F7
### Parser ###
Type: D0
Lenght: 0
 D0 38 F7 D1 38
Channel: 2
-New MSG-
96 F8 D2 38 FA D0 69 FA D1 69

-New MSG-
96 FB D2 69 FD D0 7F FD
### Parser ###
Type: D0
 D1 7F 
Lenght: 0
Channel: 3

-New MSG-
96 FE D2 7F

### Parser ###
Type: D0
Lenght: 0
Channel: 1

### Parser ###
Type: D0
Lenght: 0
Channel: 2

### Parser ###
Type: D0
Lenght: 0
Channel: 3

### Parser ###
Type: D0
Lenght: 0
Channel: 1

### Parser ###
Type: D0
Lenght: 0
Channel: 2

### Parser ###
Type: D0
Lenght: 0
Channel: 3

### Parser ###
Type: D0
Lenght: 0
Channel: 1

### Parser ###
Type: D0
Lenght: 0
Channel: 2

### Parser ###
Type: D0
Lenght: 0
Channel: 3

-New MSG-
9E A8 80 28 0

-New MSG-
9E A9 81 28 0 AA 82 28 0

-New MSG-
9F B4 80 29 1 B4 81 29 1 B5 82 29 1

-New MSG-
9F BF D0 64 BF D1 64 C0 D2 64

-New MSG-
9F C2 D0 14 C3 D1 14 C3 D2 14

-New MSG-
9F C5 80 2B A C5 81 2B A C6 82 2B A C6 D0 0 C7 D1 0

-New MSG-
9F C7 D2 0

-New MSG-
A0 91 80 35 30 92 81 35 30 92 82 35 30

-New MSG-
A0 97 80 32 2D 98 81 32 2D

-New MSG-
A0 98 82 32 2D

-New MSG-
A0 9D 80 34 3A 9D 81 34 3A 9E 82 34 3A

-New MSG-
A0 CF 80 3C 0 D0 81 3C 0 D0 82 3C 0 D1 80 37 E

-New MSG-
A0 D1 81 37 E D2 82 37 E

-New MSG-
A0 E9 80 3B 2 EA 81 3B 2 EA 82 3B 2

-New MSG-
A1 82 80 39 1 82 81 39 1 83 82 39 1

-New MSG-
A7 FD 90 47 41 FE 91 47 41 FE 92 47 41 FF 90 45 39

-New MSG-
A7 FF 91 45 39

### Parser ###
Type: F0
Lenght: 128
Channel: 0

-New MSG-
A8 80 92 45 39

-New MSG-
A9 C8 80 47 36 C9 81 47 36 CA 82 47 36

-New MSG-
A9 CC 80 45 2E CC 81 45 2E CD 82 45 2E

-New MSG-
AE E7 90 45 D E7 91 45 D E8 92 45 D

-New MSG-
AF 8C 90 47 19

-New MSG-
AF 8C 91 47 19 8D 92 47 19

-New MSG-
B0 F4 80 47 16 F4 81 47 16 F5 82 47 16

-New MSG-
B1 88 80 45 23 89 81 45 23 89 82 45 23

-New MSG-
BA EA 90 43 1E EA 91 43 1E EB 92 43 1E

-New MSG-
BC D0 80 43 1E D0 81 43 1E D1 82 43 1E

-New MSG-
83 B9 90 41 13

-New MSG-
83 BA 91 41 13 BA 92 41 13

-New MSG-
85 B0 80 41 1A B0 81 41 1A B1 82 41 1A

-New MSG-
8A B4 90 40 18 B4 91 40 18

-New MSG-
8A B5 92 40 18

-New MSG-
8A B8 90 3E 22 B9 91 3E 22 B9 92 3E 22

-New MSG-
8C 97 80 3E 1E 97 81 3E 1E 98 82 3E 1E

### Parser ###
Type: F0
Lenght: 128
Channel: 0

-New MSG-
8C A2 80 40 1C A4 81 40 1C

-New MSG-
8C A4 82 40 1C

-New MSG-
90 AF 90 3E A

-New MSG-
90 AF 91 3E A B0 92 3E A

-New MSG-
92 DD 80 3E 16

-New MSG-
92 DD 81 3E 16 DE 82 3E 16

-New MSG-
9B 93 90 3E 12 94 91 3E 12 94 92 3E 12

-New MSG-
9B AD 90 40 19

-New MSG-
9B AD 91 40 19 AE 92 40 19

-New MSG-
A6 CB 80 40 1F

-New MSG-
A6 CC 81 40 1F CC 82 40 1F

-New MSG-
A6 E1 80 3E 25 E2 81 3E 25 E3 82 3E 25

-New MSG-
AC 80 90 3E 11 81 91 3E 11

-New MSG-
AC 81 92 3E 11

-New MSG-
AE CE 80 3E 23 CF 81 3E 23 CF 82 3E 23

-New MSG-
B1 A1 90 3E F A1 91 3E F A2 92 3E F

-New MSG-
B1 AF 90 40 F AF 91 40 F B0 92 40 F

-New MSG-
B4 9E D0 A

-New MSG-
B4 9E D1 A 9F D2 A A1 D0 30

-New MSG-
B4 A1 D1 30 A2 D2 30

-New MSG-
B4 A5 D0 5E A5 D1 5E A6 D2 5E A7 D0 7B

-New MSG-
B4 A7 D1 7B A8 D2 7B AA D0 7F

-New MSG-
B4 AA D1 7F AB D2 7F

### Parser ###
Type: F0
Lenght: 128
Channel: 0

-New MSG-
B5 C7 D0 73 C7 D1 73 C8 D2 73

-New MSG-
B5 CA D0 33 CA D1 33 CB D2 33

-New MSG-
B5 CD D0 8 CD D1 8 CE D2 8

-New MSG-
B5 D0 D0 0 D0 D1 0 D1 D2 0

-New MSG-
B6 E6 D0 3 E6 D1 3 E7 D2 3

-New MSG-
B6 E9 D0 1A E9 D1 1A EA D2 1A

-New MSG-
B6 EC D0 40 EC D1 40 ED D2 40

-New MSG-
B6 EF D0 60 EF D1 60 F0 D2 60

-New MSG-
B6 F2 D0 7F F2 D1 7F F3 D2 7F

-New MSG-
B7 DC D0 5C DC D1 5C DD D2 5C

-New MSG-
B7 DF D0 25 DF D1 25 E0 D2 25

-New MSG-
B7 E2 D0 6 E2 D1 6 E3 D2 6

-New MSG-
B7 E5 D0 0 E5 D1 0 E6 D2 0

-New MSG-
B8 EF 80 40 22 EF 81 40 22 F0 82 40 22

-New MSG-
B8 FC 80 3E 22 FD 81 3E 22

-New MSG-
B8 FD 82 3E 22

-New MSG-
83 9E 90 3E 25

-New MSG-
83 9E 91 3E 25 9F 92 3E 25

-New MSG-
86 AC 80 3E 11 AC 81 3E 11 AD 82 3E 11

```

RobertoHE, 2021-08-02:
Hi @lathoub 

I think that I have found why `receive()` parser some times bugs.

I think that it is relate with the `line 238` and when a package contain 2 or more events and the middle-timeStamp is a `F7`.

https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L236-L239

Let me to put an example:
Example MSG: `08 F7 80 34 2B F7 81 34 2B F8 82 34 2B` (the same example message that last comment) 
Decoder:
`08 F7` -> TimeStamp -> Must be removed
`80` -> Note Off Ch1
`34` -> Note
`2B` -> Vel
`F7` -> TimeStamp -> MUST BE IGNORED -> `It is not a SysEx end`
`81` -> Note Off Ch2
`34` -> Note
`2B` -> Vel
`F8` -> TimeStamp -> MUST BE IGNORED
`82` -> Note Off Ch3
`34` -> Note
`2B` -> Vel

In the `receive()` algorithm, 2 first bytes that cointains timeStamp are removed. 
`leftPointer` is set in the 3rd position of buffer 0x80 (`buffer[2]`).
https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L223-L224

`while()` (line 236) puts the `rightPointer` in 5th position, 0x2B
Unfortunately, the middle-timeStamp is a `0xF7` (it may be possible if a package constens 2 o more MIDI events) and `if()` from line 238 moves the `rightPointer` one position beyond, to the 6th position.
In that point, the message must be 80, 34, 26 `(noteOff event`) but `line 238` added one byte more, and the message is transformed to 80, 34, 26, F7.

https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L236-L239

The size-filter interprete the event has 4 byte lenght (else line 251).
`Switch()` interpretes that message is a multiple 0x80 event type (noteOff multiple instead a single noteOff)
`for()` loop adds in the first iteration 3 bytes (`80, 34, 26`) and in the second one adds other 3 bytes (`80, F7*, 81*`)

*You can see how a `noteOff` can be a pitch or a velocity over 127 wrongly. Or how NoteOn or other handler may be triggered when they shouldn't.

https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L240-L284

After that, `leftPointer` is moved to 8th position (0x34). You can see that it is the second byte of the second event in the package, it skiped 0x81 (`noteOff ch2 status`).

`while (true)` loop starts again at the begining and the value of `leftPointer` is evaluated.
https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L227-L232
Now, `leftPointer` in 8th position (0x34) has a value lesser than 0x80, and it triggers the `return` prematurely. The 3rd event of the package is skiped completely. 

For this package (`08 F7 80 34 2B F7 81 34 2B F8 82 34 2B`), `receive()` had to add to the buffer the follow bytes:
`(80 34 2B / 81 34 2B / 82 34 2B`).
Instead, It added:
(`80 34 2B / 80 F7 81`)

When `MIDI.read()` will try to interprete this messenge... it will do some strange.

https://github.com/TheKikGen/midiXparser -> This midi parser works byte byte and I haven't see the same bug for the same input if you removed the first 2 bytes. It may help you how to the parser must discriminate correctly a SysEx and don't add a timeStamp F7 to buffer.

By the way, the semaphore protection may still necessary.

lathoub, 2021-08-02:
Great digging @RobertoHE 
I pulled parts of the code in MSVC and was able to get something fairly stable, just be removing line 238

So remove:
https://github.com/lathoub/Arduino-BLE-MIDI/blob/aa0f6bc44aa59995fc91493a046959619efa9df5/src/BLEMIDI_Transport.h#L238

Somehow there was never full support for SysEx across multiple packets, i'll look into that right now.

lathoub, 2021-08-02:
> By the way, the semaphore protection may still necessary.

Reading up on `xQueueSend `and `xQueueReceive `(and how they are harvested), i'm hopefull that we can do without (Branch deleted)

lathoub, 2021-08-02:
`For` this package (`08 F7 80 34 2B F7 81 34 2B F8 82 34 2B`), receive() had to add to the buffer the

The first byte (`0x08`) is not valid, bit 7 needs to be 1 (I guess a typo) (the message from the logs had the bit set to 1)

RobertoHE, 2021-08-02:
> `For` this package (`08 F7 80 34 2B F7 81 34 2B F8 82 34 2B`), receive() had to add to the buffer the
> 
> The first byte (`0x08`) is not valid, bit 7 needs to be 1 (I guess a typo) (the message from the logs had the bit set to 1)

It is true. First byte is wrong in my example. It is possible that when I copied this I accidentally changed 0x80 to 0x08. First bit always must be 1 in high timeStamp byte

RobertoHE, 2021-08-02:
> Great digging @RobertoHE
> I pulled parts of the code in MSVC and was able to get something fairly stable, just be removing line 238
>
> So remove: [...]

I have commented this line and I couldn't replicate the bug. This is good. 

I don't know how this can affect to a SysEx message splitted in any package. 

If you need some help I would investigate or try code something.

Please, review my pull request of Client #30. Don't accept it now, it needs 2 corrections (uxQueueSpacesAvailable() is useless) but I will appreciate your feedback so much.

lathoub, 2021-08-02:
> If you need some help I would investigate or try code something.

Yes, will let you know :-)

> Please, review my pull request of Client #30. Don't accept it now, it needs 2 corrections

Wilco

lathoub, 2021-08-02:
@RobertoHE 

I fixed the initial bug and initial support for (MultiPart) SysEx
https://github.com/lathoub/Arduino-BLE-MIDI/tree/MultiPartSysEx

Can you test?

I used these test sets:

```cpp
    byte sysEx1Packet[] = { 0xB0, 0xF4, 0xF0, 0x08, 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f, 0xF4, 0xF7 };
   receive(sysEx1Packet, sizeof(sysEx1Packet));

    std::cout << std::endl << "SysEx part 1" << std::endl;

    byte sysExPart1[] = { 0xB0, 0xF4, 0xF0, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08 };
    receive(sysExPart1, sizeof(sysExPart1));

    std::cout << std::endl << "SysEx part 2" << std::endl;

    byte sysExPart2[] = { 0xB0, 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f, 0xF4, 0xF7 };
    receive(sysExPart2, sizeof(sysExPart2));

    std::cout << "ble Packet with 1 MIDI messages" << std::endl;

    byte blePacketWithOneMIDIMessage[] = { 0xA8, 0xC0, 0x90, 0x3E, 0x3E };
    receive(blePacketWithOneMIDIMessage, sizeof(blePacketWithOneMIDIMessage));

    std::cout << std::endl << "ble Packet with 2 MIDI messages" << std::endl;

    byte blePacketWithTwoMIDIMessage[] = { 0xA8, 0xC0, 0x90, 0x3E, 0x3E, 0xC1, 0x91, 0x3E, 0x3E };
    receive(blePacketWithTwoMIDIMessage, sizeof(blePacketWithTwoMIDIMessage));

    std::cout << std::endl << "2 MIDI messages with running status" << std::endl;

    byte twoMIDIMessageWithRunningStatus[] = { 0xA9, 0xAE, 0xD1, 0x74, 0xAF, 0xD2, 0x74, 0xB1 };
    receive(twoMIDIMessageWithRunningStatus, sizeof(twoMIDIMessageWithRunningStatus));

    std::cout << std::endl << "2 MIDI messages with running status" << std::endl;

    byte multipleMIDIMessagesMixedType[] = { 0x00 };
    receive(multipleMIDIMessagesMixedType, sizeof(multipleMIDIMessagesMixedType));

```

RobertoHE, 2021-08-03:
Hi @lathoub 
> Can you test?
> 
I will try to test it this evening. I may add some 0xF0 to 0xFF events to complete the test, if I have time.

> I fixed the initial bug and initial support for (MultiPart) SysEx
> https://github.com/lathoub/Arduino-BLE-MIDI/tree/MultiPartSysEx
>

I have some doubts:
`lPtr` starts to evaluate the buffer `(timeStampHigh`) at the second byte (`buffer[1]`) instead at the beginning (`buffer[0]`)
line 233 must be after 237 line and line 239 must be after line 241.
https://github.com/lathoub/Arduino-BLE-MIDI/blob/802f7a3c9c3f7b76b02a4fe06e0074553c49bd17/src/BLEMIDI_Transport.h#L224-L242

`rPtr` increases twice. Line 345 is unnecessary. In this part of the code, it evaluates the `timeStampLow` as event divider, which has  only 1 byte lenght, not 2. If it increases twice, I will evaluate the next `midiStatus` instead `timeStampLow divider`.

Lines 353-356 may be redundant only if `rPrt` increase once. It may be considered like redundant protection, it is not bad.
https://github.com/lathoub/Arduino-BLE-MIDI/blob/802f7a3c9c3f7b76b02a4fe06e0074553c49bd17/src/BLEMIDI_Transport.h#L340-L356

Lines 321 to 332 are duplicated inside and afer `switch()`, the code does the same twice.` rPtr` avances twice, the first one inside of `switch()` and the second one after `switch()`. It may evaluate the next `midiStatus` instead `timeStamp`, if it increase twice.
https://github.com/lathoub/Arduino-BLE-MIDI/blob/802f7a3c9c3f7b76b02a4fe06e0074553c49bd17/src/BLEMIDI_Transport.h#L316-L334

What do you opine about these corrections?

lathoub, 2021-08-03:
@RobertoHE very well spotted Sir! (I was trying to prematurely anticipating the removing of the timestamp statements)
Correction on its way

RobertoHE, 2021-08-03:
Hi again

Are you sure that these lines are necessary?

> Lines 321 to 332 are duplicated inside and afer `switch()`, the code does the same twice.` rPtr` avances twice, the first one inside of `switch()` and the second one after `switch()`. It may [...]

I don't know if after a splitted SysEx message can appear other event in the same package. In that case, it moves `rPrt` twice wrongly beyond `midiStatus` of the next event.

I let me put an example (I don't know if it possible):

Package 1: `80 F7 80 34 2B F7 81 34 2B F8 **F0** 00 11`
Package 2: `91 A2 22 33 44 55 **F7** A3 90 34 90 A4 80 34 00`

PCK1:
80 F7 -> TimeStamp High and Low
80 -> NoteOff Ch 1
34 -> pitch
2B -> Velocity 
F7 -> TimeStamp Low (divider) 
81 -> NoteOff ch2
34 -> pitch
2B -> Velocity 
F8 -> TimeStamp Low (divider)
**F0** -> SysEx Start
11 22 -> SysEx Data

PCK2:
91 A2 -> TimeStamp High and Low
33 44 55 -> SysEx Data
**F7** -> SysEx End
A3 -> TimeStamp Low (divider)
90 -> noteOn Ch 0
34 -> pitch
90 -> velocity
A4 ->TimeStamp Low (divider)
80 -> noteOff Ch 0
34 -> pitch
90 -> velocity

PCK1 set the flag `sysExContinuation` because it conains F0 but not F7.

PCK2 is processed correctly until F7 as SysEx. `switch()` go to `case MIDI_NAMESPACE::MidiType::SystemExclusive:` and `for()` add the rest of message to buffer. Until now, everything is correct. rPtr points to F7 (buffer[6])

Inside `case MIDI_NAMESPACE::MidiType::SystemExclusive:`, `rPtr` is increased (`buffer[7]` = A3 TimeStampLow) by lines 319-321 and parser checks that package is completely scanned. In the example, that is `false`, and parse continue.

In lines 323-327, it captures the `timeStampLow`  (`buffer[7]` = A3) and after it increases `rPtr` to (`buffer[8]` -> `midiStatus` 90 `noteOn`). 

**In line 329, `rPtr` is moved again to `buffer[9]`(why?) (`pitch` of `noteOn` 34). In this point, it have skipped the `midiStatus`**. After that, `switch()` gets exit and parser continue at line 336.

**Lines 337 to 346 are the same that lines 319 to 327 and parser does twice the same for a SysEx.**

At the line 337, `rPtr` is increased again and now it points to `buffer[10]` (`vel` of `noteOn`). It is not the end of the message, and the parser continue to line 341.
At the line 341, parser capture `buffer[10]` (`vel `of `noteOn`) like timeStampLow and move the point to `buffer[11]` (`timeStampLow`).
After that. `while()` loop ends and the parser starts again with `lPtr` in `buffer[11]` (`timeStampLow` A4) and it miss-interpretes it like a midiStatus. .... I don't continue explaining, but you can suppose what happen after that....

In resume, it miss-interpreted everything after F7 SysEx end (NoteOn and NoteOff events).

https://github.com/lathoub/Arduino-BLE-MIDI/blob/687676dc44042809ac1b21faafec2677e89d9c30/src/BLEMIDI_Transport.h#L314-L352

If you repeat the sequence with the 2 packages and you remove the lines 319-329, you will can see that message will be processed correctly.

Would you check it, please?

lathoub, 2021-08-03:
No sure if PCK1 and PCK2 are valid according to the spec - i need to further check.

Here is what the spec says for (multipacket) SysEx:
[image]

lathoub, 2021-08-03:
I fixed an additional bug (test for SysExEnd) and added MSVC parser test code

lathoub, 2021-08-04:
Further cleanup and testing (in MSVC, not in the Arduino) and cleared all of the test data - getting closer :-)
(Including fixing/cleaning some of the remarks made above)

RobertoHE, 2021-08-04:
Please, check #32.

lathoub, 2021-08-06:
Changes accepted - thanks you @RobertoHE

lathoub, 2021-08-06:
https://github.com/lathoub/Arduino-BLE-MIDI/blob/c20062ec6322ffff48d861511fedd906c5e2789c/src/BLEMIDI_Transport.h#L229-L231

You recommend **not** using `RUNNING_ENABLE` - yet the underlying parser does support runningStatus for incoming messages and runs fine with the `RUNNING_ENABLE`  enabled or disabled (running msvc examples renders the same result). I kinda like the efficiency that runningStatus brings.

Based on the current examples in BLEMIDI_Sim.h, i recommend making `RUNNING_ENABLE` permanent. WDYT @RobertoHE ?

https://github.com/lathoub/Arduino-BLE-MIDI/blob/c20062ec6322ffff48d861511fedd906c5e2789c/test/msvc/BLEMIDI_Sim.h#L106-L110

lathoub, 2021-08-06:
https://github.com/lathoub/Arduino-BLE-MIDI/blob/c20062ec6322ffff48d861511fedd906c5e2789c/src/BLEMIDI_Transport.h#L341-L342

Is this code needed? Looks like it can be removed. The switch below does not test for `SystemExclusive`

RobertoHE, 2021-08-06:
You are right, it is not neccessary.

0xF0s have 1 byte lenght, except SysEx.

> https://github.com/lathoub/Arduino-BLE-MIDI/blob/c20062ec6322ffff48d861511fedd906c5e2789c/src/BLEMIDI_Transport.h#L341-L342
> 
> Is this code needed? Looks like it can be removed. The switch below does not test for `SystemExclusive`

RobertoHE, 2021-08-06:
> https://github.com/lathoub/Arduino-BLE-MIDI/blob/c20062ec6322ffff48d861511fedd906c5e2789c/src/BLEMIDI_Transport.h#L229-L231
>
> You recommend **not** using `RUNNING_ENABLE` - yet the underlying parser [...]

I only have tested the parser with default settings of serial parser of main midi class. I haven't have time for check if any other options of setting class would interference with runningStatus mode.

If any setting option not interferences with it, runningStatus enable is prefered because it uses less resorces.

Other option is change #defines by if(Settings::option). This will add some computer cycles to parser but it would be transparent to end user or end application.

Would you check it in paralle?

lathoub, 2021-08-07:
Parser is going OK on an ESP32 Devkit v1 using the default ESP32 BLE stack - I tried all kinds on MIDI messages and all come through - good stuff. I'm good to commit to master!

FYI: I also tried the NimBLE stack and also ran OK, but I noticed that when Disconnecting, the connect button on the Apple MIDI setup does not come back and the device disappears.

RobertoHE, 2021-08-08:
> 
> FYI: I also tried the NimBLE stack and it also ran OK, but I noticed that when Disconnecting, the connect button on the Apple MIDI setup does not come back and the device disappears.

I have never used Apple MIDI. I don't know anything about it. What is the problem? What is the setup?

lathoub, 2021-08-09:
> I have never used Apple MIDI. I don't know anything about it. What is the problem? What is the setup?

oho - you see, i can't do 2 things at the same time: work on [AppleMIDI ](https://github.com/lathoub/Arduino-AppleMIDI-Library)and BLE-MIDI :-).

What i was trying to say: the ESP32 connects fine to the Apple BLE stack, but after disconnect, the ESP32 does not re-appear and i can no longer connect. Does it behave correctly with you?

RobertoHE, 2021-08-10:
Hi again

Client mode doesn't use advertiments, it scans the advertised devices and it connect to them.

When I was coding the client class, I used like template the nimble server of this repo and the nimble client example. I noticed that nimble example allowed multi-client connection and it adverted in loop. Your nimble server only advertice in setup and it doesn't do nothing special after a disconnection.

Client mode has an auto-recconect mechanism inside read function that it is managed with onConnect and onDisconnect callbacks. For a server

RobertoHE, 2021-08-17:
> What i was trying to say: the ESP32 connects fine to the Apple BLE stack, but after disconnect, the ESP32 does not re-appear and i can no longer connect. Does it behave correctly with you?

Hi @lathoub . Check this function:
https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_server.html#a6bfd923ecd0ea06d5564343ab7209122

lathoub, 2021-09-22:
> Hi @lathoub . Check this function:
> https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_server.html#a6bfd923ecd0ea06d5564343ab7209122

Fixed!

lathoub, 2021-09-23:
All code moved to the master branch. Thank @RobertoHE 
(I activated the RUNNING_ENABLE flag, as the underlying parser uses running status)

closed 2021-09-23 by lathoub