lathoub/Arduino-BLE-MIDI #31: No BLE advertisement when using example sketch

joeybab3, 2021-07-14:
I may just be missing something obvious but when running the example sketch on the ESP32 dev board I don't see any device advertised. 

I do see a device advertised when using the NIM-BLE examples so I don't think it's an issue with my board.

Any ideas?

joeybab3, 2021-07-15:
Interestingly enough it does appear to be advertising but it does not show up on any device I use to try and connect. 

Using nRF connect I can see the device advertising though. Does anyone know what might be causing this?

lathoub, 2021-09-23:
I updated the code the re-advertise after disconnect - can you try again?

lathoub, 2021-12-09:
@joeybab3 pls check

Theo428, 2022-01-15:
I am having the same issue with a nodemcu-32s board. I just downloaded the latest version of the library using library manager. I do not have a way to monitor if the device is broadcasting or not like the nRF mentioned above but it does advertise with the nimBLE example server but does not with the example sketches provided. I don't see it on either windows 11 or android if that makes any difference. I don't have any other devices to test on.

lathoub, 2022-01-17:
Can you try by downloading from this site (not using the library manager, but download the zip file). I was awaiting the feedback from @joeybab3 before I publish to the Arduino catalog

Theo428, 2022-01-18:
> Can you try by downloading from this site (not using the library manager, but download the zip file). I was awaiting the feedback from @joeybab3 before I publish to the Arduino catalog

It appears to work now. I can see the device in both windows and on my phone and I connect and receive the data.

joeybab3, 2022-01-18:
> 

I will try it asap, I am in school now so I don't have as easy access to my development stuff

lathoub, 2022-02-06:
paging @joeybab3

lathoub, 2022-02-21:
@joeybab3 last call

joeybab3, 2022-02-22:
I still don't have anything to test it on at the moment as I am in school. Feel free to close the issue and I will reopen it if I can test it at a later time and find that the issue is still existing.

closed 2022-02-23 by lathoub