lathoub/Arduino-BLE-MIDI #40: BLE-MIDI Client: Problems when reconnecting to a MIDI server

MicroMidi, 2021-12-13:
First of all I'd like to thank so much for your brilliant library - it works (almost) flawlessly as a base for a MIDI footswitch for my guitar effect devices.
The “Nux Mighty Plug” works without any problems with your library – it uses standard MIDI-CC messages for effect swithing. But a a strange "phenomena" occurs with my VOX Adio amp. This amp only works with MIDI SysEx messages - and needs a delay of 400ms after the Bluetooth connection is established and before the first MIDI message could be sent, probably for internal initialization purposes. I implemented this delay in the “OnConnect”-callback routine. Everything works fine when the amp is already switched on and the ESP32 client with your library gets started. When I switch off the amp the BLE-connection gets lost and the client starts scanning for other BLE-MIDI devices. When I turn on the amp again the client reconnects, but the amp doesn't respond to MIDI messages. Everything works fine again when I reset the ESP32-client and the connection gets re-established from scratch.
I recognized that your BLE-Client code displays a "Connected to"-message in combination with the name and MAC-address of the MIDI-Server in an intitial connection scenario. In a reconnect-scenario only the "Connected to"-message appears in serial monitor. Is a re-connect scenario treated different from an initial connection in your library and is the something special to consider? I’m quite sure that the amp in general is a very sensitive MIDI server and your library is not the cause for the re- connection problems … but probably you have a hint for me to solve my problem - thanks in advance!

lathoub, 2021-12-15:
Hi @MicroMidi thanks for the question.
Quick question before deep diving into the issue: do you get a disconnect event?

lathoub, 2021-12-17:
I'm going to bring in @RobertoHE who wrote the client part of the library, maybe he sees what is happening

lathoub, 2021-12-17:
Can you alter:

https://github.com/lathoub/Arduino-BLE-MIDI/blob/5c42e1470451008de2aed085fdff9bf0bbd60f79/src/hardware/BLEMIDI_Client_ESP32.h#L333-L346

with 

```cpp
    void onDisconnect(BLEClient *pClient)
    {
        if (_bluetoothEsp32)
        {
            _bluetoothEsp32->disconnected();
            // Try reconnection or search a new one
            _bluetoothEsp32->scan();
        }
    }
```

I can't test it on my side, but worth trying

MicroMidi, 2021-12-19:
Thanks again for your support @lathoub! I tried your suggestion - but I didn't solve my problem. Nevertheless the onDisconnect() method was a good hint to try out some other approaches myself ... and what finally did the trick for me was to delete and create the NimBLEDevice-client. Now it looks like the client is re-instantiated from scratch after a re-connect and everything works now perfectly.

This is the current code I use with the added "Renew Client" code:

**void onDisconnect(BLEClient *pClient)
{
  if (_bluetoothEsp32)
  {
    _bluetoothEsp32->disconnected();
    // Try reconnection or search a new one
    _bluetoothEsp32->scan();
  }

  // Renew Client
  NimBLEDevice::deleteClient(pClient);
  NimBLEDevice::createClient();

  //Try reconnection or search a new one
  NimBLEDevice::getScan()->start(1, scanEndedCB);
}**

I guess that this is not a very smart approach because other MIDI devices to not need the "hard reset" while reconneting. But for me it works without any problems.

I will try now to build a hybrid client for my "Nux Mighty Plug" and "Adio Air GT" devices. Is there a way to figure out the name of the connected device in the client code in order to distinguish what MIDI messages to send to the external device?

Thanks in advance
Bernd

lathoub, 2021-12-20:
Great - happy you were to able to fix it.
Please create a PR, so i can review the changes and add it to the code-base

MicroMidi, 2021-12-20:
I'm quite new to GitHub - so I had to google for the term "PR" - it seems to be a "Pull Request" where I fork your code into my own repository and adopt my changes in that environment, right? I will try to make this happen, cross your fingers for me;-)

Thanks and greets, Bernd

MicroMidi, 2021-12-20:
... I created a PR with the modified code. Can you please have a look if everything was prepared as expected?

lathoub, 2021-12-20:
> I'm quite new to GitHub - so I had to google for the term "PR" - it seems to be a "Pull Request" where I fork your code into my own repository and adopt my changes in that environment, right? I will try to make this happen, cross your fingers for me;-)

:-) learn something new every day 👍

lathoub, 2021-12-20:
The PR has not yet shown up in this repo. I did see it in your [repo](https://github.com/MicroMidi/Arduino-BLE-MIDI/pull/1), now bring it over to this repo

RobertoHE, 2021-12-20:
Hi @MicroMidi and @lathoub 

Be carefull with this change (it may be a correct solution, of course):

> This is the current code I use with the added "Renew Client" code snippet:
> 
> void onDisconnect(BLEClient *pClient) { if (_bluetoothEsp32) { _bluetoothEsp32->disconnected(); // Try reconnection or search a new one _bluetoothEsp32->scan(); }
> 
> // Renew Client NimBLEDevice::deleteClient(pClient); NimBLEDevice::createClient();
> 
> //Try reconnection or search a new one NimBLEDevice::getScan()->start(1, scanEndedCB); }
> 
> I guess that this is not a very smart approach because other MIDI devices may not need this kind of "hard reset" while reconneting. But for me it works without any problems.
> 

It affects to the next part of the code:

https://github.com/lathoub/Arduino-BLE-MIDI/blob/5c42e1470451008de2aed085fdff9bf0bbd60f79/src/hardware/BLEMIDI_Client_ESP32.h#L484-L514

These lines of code try to reconnect the device using the previous info of the device. This method is recommended by nimBLE lib because (theoretically) saves time and power because skips send some initial config messages. 

Your solution may be valid too, but it may cause some little delays after a reconnection because need a little bit more time than a reconnection. They may be inappreciable but it is a thing to consider.

If the solution that propose @MicroMidi is accepted for a merge, maybe the lines that I put before may be removed from the code because the client never can perform a reconnection because the client will removed after a disconnection,

RobertoHE, 2021-12-20:
And it may be recommended add `_client = nullptr;` too after `NimBLEDevice::deleteClient(pClient); NimBLEDevice::createClient();`

MicroMidi, 2021-12-21:
@RobertoHE: Thanks for the clarification ... now I understand another concept of your BLE MIDI Client code. To me it seems that my proposed code change enforces that the BLE connection will always be built up from scratch - and this makes the part of the code that you mentioned in your previous post obvious. I guess that 99% of all MIDI servers can deal with the reconnection approach without any problems - so we definitely

RobertoHE, 2021-12-21:
Hi @MicroMidi .

Let me some days for implemets the changes.

MicroMidi, 2021-12-22:
Thanks @RobertoHE , no need to hurry - please take all the time you need!

RobertoHE, 2021-12-23:
@MicroMidi 
Check the proposed changes in this fork https://github.com/RobertoHE/Arduino-BLE-MIDI before that I will do a pull request to official repo.

MicroMidi, 2021-12-27:
@RobertoHE
Thanks for your time and effort! When I use your updated library then the bluetooth scan stops after disconnecting the bluetooth server. The scan runs exactly once after the disconnect happens and then the scan stops. I haven't found out why - but it happens identically with two different amps that I tested as bluetooth server...

RobertoHE, 2022-01-18:
Hi again @MicroMidi 
Sorry for the delay.

I added your changes to code.

Would you check the repo again? Uncomment this line before compile https://github.com/RobertoHE/Arduino-BLE-MIDI/blob/ebe18daaea8b1cea0826bbeae893402d12b73f2c/src/hardware/BLEMIDI_Client_ESP32.h#L132

If it works, I will pull request the changes to main repo.

MicroMidi, 2022-01-19:
Dear Roberto,

there was no need to hurry – thanks again for your time and effort.

Everything works perfectly now! The parameter to force a new BLE-client after a reconnect does the trick for my Vox Amp – really cool.

Thank you so much for incorporating my kind of exotic functional request into the official library code, kind regards

Bernd aka MicroMidi

Von: RobertoHE
Gesendet: Dienstag, 18. Januar 2022 18:11
An: lathoub/Arduino-BLE-MIDI
Cc: MicroMidi; Mention
Betreff: Re: [lathoub/Arduino-BLE-MIDI] BLE-MIDI Client: Problems whenreconnecting to a MIDI server (Issue #40)

Hi again @MicroMidi
Sorry for the delay.
I added your changes to code.
Would you check the repo again? Uncomment this line before compile https://github.com/RobertoHE/Arduino-BLE-MIDI/blob/ebe18daaea8b1cea0826bbeae893402d12b73f2c/src/hardware/BLEMIDI_Client_ESP32.h#L132
If it works, I will pull request the changes to main repo.

RobertoHE, 2022-01-19:
I have proposed a pull request, @lathoub.
Would you check and merge it to the main repo?
Thanks.

lathoub, 2022-01-19:
@RobertoHE thank you for the work, looks like it works - yeah! Checked and merged.
(thank you @MicroMidi for raising the issue and testing)

With this done, I will create a new release (and make available in the Arduino library)

closed 2022-01-19 by lathoub

MicroMidi, 2022-01-19:
@lathoub @RobertoHE : This is pretty perfect - thanks for incorporating my request in an official release – it works really great and reliable now!