lathoub/Arduino-BLE-MIDI #50: ESP32 compatibility MIDI.send

ChristopheLebreton, 2022-02-17:
Hi,

I just tested your lib with different ESP32 and compared also performances from other BLE midi lib.
yours is actually the best, I like it ;)
but unfortunately I have a problem with Wemos D1 mini ESP32 model :
no midi sent doesn't work and I don't find yet why ?
I used MidiBle example with #include <hardware/BLEMIDI_ESP32.h>
Exactly the same code setting and test with a ESP32 Dev Kit C work perfectly.

If you have any idea why and how we have to do to resolve this problem it'll be perfect ;)

all the best, and thx for your lib :))

lathoub, 2022-02-17:
Were you able to solve the issue @ChristopheLebreton ? What was the cause? (so that folks with similar hardware can learn from your findings)

ChristopheLebreton, 2022-02-17:
Yes it’s solved.
I supposed when I uploaded many time, different code with the same name of BLEMIDI instance, OSX Monterey don’t like it and created strange issue.
So after a morning reboot everything it’s ok :)
I froward to you if I’ll meet some another problem.

Actually, I worked specifically with callback to received many controlChange dumps to be read and parse for some treatments inside void loop().
The difficulty is to missed some messages between the callback and loop(). Certainly a problem of interruption and cache between BLE core service and other one used for loop() !?
To solve it, I’ve to repeat sent of all controlChange to have a maximum probability they arrived inside loop().
If you have a better idea to solve kind of these problem, I'm sure it'll be great for the future ;)

best regards :))

lathoub, 2022-02-17:
> The difficulty is to missed some messages between the callback and loop(). Certainly a problem of interruption and cache between BLE core service and other one used for loop() !?

Can you post a minimal example for me to reproduce?

> Certainly a problem of interruption and cache between BLE core service and other one used for loop() !?

Not sure, `xQueue*` commands are supposed to take care of this

https://github.com/lathoub/Arduino-BLE-MIDI/blob/8dee1f7d257aa71aeb0e956632da2dd2ae9ed496/src/hardware/BLEMIDI_ESP32_NimBLE.h#L43

https://github.com/lathoub/Arduino-BLE-MIDI/blob/8dee1f7d257aa71aeb0e956632da2dd2ae9ed496/src/hardware/BLEMIDI_ESP32_NimBLE.h#L49

ChristopheLebreton, 2022-02-21:
Hi,

please find a code I used to test with a MaxMSP patch to test it.
it works for me after different test.
If I used this technique inside a complex project I have to repeat midi sent message to ESP32 to be sure all are read by loop()... So it's perhaps not the best technique...

let me know if you have a better solution ;)

regards :)

[BLE-MIDI_Receiving_ControlChange_test.zip](https://github.com/lathoub/Arduino-BLE-MIDI/files/8110672/BLE-MIDI_Receiving_ControlChange_test.zip)

lathoub, 2022-02-21:
I have uploaded your code for future reference (slightly modified for readability).
Still not sure what the issue is and how you resolve it. Are messages dropped?

```cpp
#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_ESP32.h> // Tested with ESP32 DEV Kit C & WEMOS D1 mini ESP32
//#include <hardware/BLEMIDI_ESP32_NimBLE.h> // Tested with ESP32 DOIT dev Kit v1

BLEMIDI_CREATE_DEFAULT_INSTANCE()
bool isConnected = false;

uint8_t ctlNumber = 0;
uint8_t ctlValue = 0;
uint8_t ctlFlag [4] = {0, 0, 0, 0};

void setup() {
  Serial.begin(115200); // Start the serial console
  while (!Serial);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  BLEMIDI.setHandleConnected([]() {
    isConnected = true;
    digitalWrite(LED_BUILTIN, HIGH);
  });
  BLEMIDI.setHandleDisconnected([]() {
    isConnected = false;
    digitalWrite(LED_BUILTIN, LOW);
  });

  MIDI.setHandleControlChange([](midi::Channel channel, byte controller, byte value) {
    Serial.printf("Received control change : channel %d, controller %d, value %d \n", channel, controller, value);
    ctlNumber = controller;
    ctlValue = value;
  });

  MIDI.begin(MIDI_CHANNEL_OMNI);
}

void loop() {
  MIDI.read();

  if (isConnected) {
    // this expression missed fourth i !?
    for (int i = 0; i <= 3; i++) {
      if (ctlNumber == i + 1 && ctlValue != ctlFlag[i]) {
        MIDI.sendControlChange(i + 1, ctlValue, 1);
        ctlFlag[i] = ctlValue;
        Serial.printf("Received ctl_1 OK : %d %d \n", i, ctlValue);
      }
    }
  }
}
```

I suggest to modify the loop() , but not sure if this address the issue.

```cpp
void loop() {
  if (isConnected) {
    if (MIDI.read()) {
      // this expression missed fourth i !?
      for (int i = 0; i <= 3; i++) {
        if (ctlNumber == i + 1 && ctlValue != ctlFlag[i]) {
          MIDI.sendControlChange(i + 1, ctlValue, 1);
          ctlFlag[i] = ctlValue;
          Serial.printf("Received ctl_1 OK : %d %d \n", i, ctlValue);
        }
      }
    }
  }
}
```

ChristopheLebreton, 2022-02-22:
actually to solve the problem of dropped message inside a heavy project, I send to the ESP32 all ctlChange messages each 100ms. For my current project it's not a really problem because these ctlChange used only to set some parameters, I don't need real real-time.
just to confirm callback always received ctlChange, it's the link between callback -> loop() the random problem.
best ;)

closed 2022-02-22 by lathoub