lathoub/Arduino-BLE-MIDI #51: ESP32 not Advertising to most of my devices

Televisioncomputer, 2022-02-24:
I have just setup BLEmidi on my ESP32, but only my Linux Desktop can see the ESP (Cant connect to it, it just sees it), all of my other laptops/phones cant see the ESP, I have tried installing BLEmidi from the Library Manager and tried to manually install the Autumn 21 Release but both gave me the same issue

lathoub, 2022-02-24:
try downloading the source code directly from this repo
(I'm about to make a release with the latest changes)

Televisioncomputer, 2022-02-24:
Nice, now all my devices can see it now, but I still cant connect to it =/

lathoub, 2022-02-25:
> Nice, it works now, thanks

Feel free to close the issue when you ar happy with the outcome.
I will issue a release this weekend, so the latest changes are available in the lib manager.