lathoub/Arduino-BLE-MIDI #55: Can't compile after update to 2.2 / Arduino BLE

lathoub, 2022-03-24:
### Discussed in https://github.com/lathoub/Arduino-BLE-MIDI/discussions/53

<div type='discussions-op-text'>

<sup>Originally posted by **agentff6600** March 23, 2022</sup>
Hi!

I updated to the new version today. I tried compile MidiBle.ino again onto an Arduino NANO 33 IOT (that worked before the update).

Now I get this message:

```
In file included from /private/var/folders/gf/tr166wdx0cbcrcgq6kl_6h7w0000gn/T/.arduinoIDE-unsaved2022223-32034-lm06d2.6h9ve/MidiBle/MidiBle.ino:6:0:
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:5:24: error: 'SERVICE_UUID' was not declared in this scope
 BLEService midiService(SERVICE_UUID);
                        ^~~~~~~~~~~~
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:5:24: note: suggested alternative:
In file included from /private/var/folders/gf/tr166wdx0cbcrcgq6kl_6h7w0000gn/T/.arduinoIDE-unsaved2022223-32034-lm06d2.6h9ve/MidiBle/MidiBle.ino:1:0:
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:21:26: note:   'bleMidi::SERVICE_UUID'
 static const char *const SERVICE_UUID        = "03b80e5a-ede8-4b33-a751-6ce34ec4c700";
                          ^~~~~~~~~~~~
In file included from /private/var/folders/gf/tr166wdx0cbcrcgq6kl_6h7w0000gn/T/.arduinoIDE-unsaved2022223-32034-lm06d2.6h9ve/MidiBle/MidiBle.ino:6:0:
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:6:34: error: 'CHARACTERISTIC_UUID' was not declared in this scope
 BLEStringCharacteristic midiChar(CHARACTERISTIC_UUID,  // standard 16-bit characteristic UUID
                                  ^~~~~~~~~~~~~~~~~~~
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:6:34: note: suggested alternative:
In file included from /private/var/folders/gf/tr166wdx0cbcrcgq6kl_6h7w0000gn/T/.arduinoIDE-unsaved2022223-32034-lm06d2.6h9ve/MidiBle/MidiBle.ino:1:0:
/Users/windlin/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:22:26: note:   'bleMidi::CHARACTERISTIC_UUID'
 static const char *const CHARACTERISTIC_UUID = "7772e5db-3868-4112-a1a9-f2669d106bf3";
                          ^~~~~~~~~~~~~~~~~~~
```

I am not sure what to take from this? Is it pointing at a file in var that should not be there? </div>

lathoub, 2022-03-24:
It seems ArduinoBLE is no longer working :-(. 
I can do some of the work to fox it, but will need help (I no longer have the hardware) - can you help?

lathoub, 2022-03-24:
(thank you for signalling! 🙏)

lathoub, 2022-03-24:
I started a new branch to work on Nano 33 support
https://github.com/lathoub/Arduino-BLE-MIDI/tree/Arduino-Nano-BLE-33

agentff6600, 2022-03-24:
Hi!
just posted a message in the discussion but I'll add it here too.

I reverted to version 2.1.0 which seems to resolve the problem.

Then I wonder what difference there is between 2.2.0 and 2.1.0?

lathoub, 2022-03-24:
Moved `midiChar` and `midiService` globals into the `BEGIN_BLEMIDI_NAMESPACE` namespace. (the global approach is not good and must be corrected).

It now compiles, but I can't test it - can you?

agentff6600, 2022-03-25:
I can try to look into it in the afternoon. I guess I use the link above to alter my library?

tommitan, 2022-05-19:
Hi,
I'm trying to make a MIDI with Arduino (I just want to make music after some sensors conditions) to connect Arduino with my Mac or iPad. I've Arduino Nano Rp 2040 and a MKR VIDOR 4000.
At the moment I'm using the first one with this library.
With the example code:

#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_ArduinoBLE.h>

BLEMIDI_CREATE_INSTANCE("CustomName", MIDI)

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  BLEMIDI.setHandleConnected(OnConnected);
  BLEMIDI.setHandleDisconnected(OnDisconnected);
  
  MIDI.begin();
}
void loop()
{
  MIDI.read();
}

void OnConnected() {
  digitalWrite(LED_BUILTIN, HIGH);
}

void OnDisconnected() {
  digitalWrite(LED_BUILTIN, LOW);
}
   

I 've got this error message:

In file included from /Users/tommi/Documents/Arduino/cancella/cancella.ino:2:0:
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:5:24: error: 'SERVICE_UUID' was not declared in this scope
 BLEService midiService(SERVICE_UUID);
                        ^~~~~~~~~~~~
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:5:24: note: suggested alternative:
In file included from /Users/tommi/Documents/Arduino/cancella/cancella.ino:1:0:
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:21:26: note:   'bleMidi::SERVICE_UUID'
 static const char *const SERVICE_UUID        = "03b80e5a-ede8-4b33-a751-6ce34ec4c700";
                          ^~~~~~~~~~~~
In file included from /Users/tommi/Documents/Arduino/cancella/cancella.ino:2:0:
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:6:34: error: 'CHARACTERISTIC_UUID' was not declared in this scope
 BLEStringCharacteristic midiChar(CHARACTERISTIC_UUID,  // standard 16-bit characteristic UUID
                                  ^~~~~~~~~~~~~~~~~~~
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:6:34: note: suggested alternative:
In file included from /Users/tommi/Documents/Arduino/cancella/cancella.ino:1:0:
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:22:26: note:   'bleMidi::CHARACTERISTIC_UUID'
 static const char *const CHARACTERISTIC_UUID = "7772e5db-3868-4112-a1a9-f2669d106bf3";
                          ^~~~~~~~~~~~~~~~~~~
In file included from /Users/tommi/Documents/Arduino/cancella/cancella.ino:1:0:
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h: In instantiation of 'void bleMidi::BLEMIDI_Transport<T, _Settings>::receive(byte*, size_t) [with T = bleMidi::BLEMIDI_ArduinoBLE; _Settings = bleMidi::DefaultSettings; byte = unsigned char; size_t = unsigned int]':
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/hardware/BLEMIDI_ArduinoBLE.h:107:65:   required from here
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:297:60: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             while ((buffer[rPtr + 1] < MIDI_TYPE) && (rPtr < (length - 1)))
                                                      ~~~~~~^~~~~~~~~~~~~~~
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:391:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             if (++rPtr >= length)
                 ~~~~~~~^~~~~~~~~
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:408:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             if (lPtr >= length)
                 ~~~~~^~~~~~~~~
/Users/tommi/Documents/Arduino/libraries/BLE-MIDI/src/BLEMIDI_Transport.h:260:14: warning: unused variable 'timestampHigh' [-Wunused-variable]
         auto timestampHigh = 0x3f & headerByte;
              ^~~~~~~~~~~~~
exit status 1
Errore durante la compilazione per la scheda Arduino Nano RP2040 Connect.

I'm totally new in this word and a very beginner... so I've literally no idea what to do... can u help me pleaseee!!
I need a solution for my MIDI in a day or two!!

If there's no way to work this out, there's another way (an easy one...) to make sound out of an Arduino sketch???

Thanks, Tom

lathoub, 2022-05-19:
Hi @tommitan 

Use the branch for the Nano 33/rp2040 (see above)
https://github.com/lathoub/Arduino-BLE-MIDI/issues/55#issuecomment-1077437382

Let me know if it works, so I can roll it up in a release

richarnt999, 2022-06-11:
I have the same problem, and after updating the library at least it compiles. Will let you know if the hardware talks later (that might of course be due to numerous other problems - I hope to get my arduino nano ble 33 talking to a CME Widi master).

richarnt999, 2022-06-11:
> I have the same problem, and after updating the library at least it compiles. Will let you know if the hardware talks later (that might of course fail due to numerous other problems - I hope to get my arduino nano ble 33 talking to a CME Widi master).

Tested and seems to work like a charm.

lathoub, 2022-06-11:
> Tested and seems to work like a charm.

Please close issue when you are happy

richarnt999, 2022-06-11:
Close #55

SunboX, 2022-06-14:
> I started a new branch to work on Nano 33 support https://github.com/lathoub/Arduino-BLE-MIDI/tree/Arduino-Nano-BLE-33

Will this also work on a Nano RP2040 ? I did just order one.

lathoub, 2022-06-14:
Hi, yes, i have it working on a RP2040 - let me know here if it also works for you!

lathoub, 2022-06-27:
@SunboX did you get it to work on the RP2040?

spotman, 2022-07-30:
@lathoub I've just tested Arduino-Nano-BLE-33 branch on my hardware and it works as expected. May you merge these changes in the master branch?

lathoub, 2022-07-31:
thx @spotman - all changes for Arduino Nano-BLE-33 are also copied in the newer [CustomSettings branch](https://github.com/lathoub/Arduino-BLE-MIDI/tree/CustomSettings). Would you might testing that branch?

spotman, 2022-08-01:
@lathoub yep. I have an Arduino Nano 33 BLE board, apps for testing BLE-MIDI protocol and an oscilloscope. What kind of tests I need to run on this branch?

lathoub, 2022-08-01:
the same test you ran before on the BLE branch - internally I added template parameters and I want to make sure they compile and work as expected (they got rather complex and want to see if they also work on another machine then mine :-))

tarasitri, 2023-05-08:
Dear @lathoub,
Thank you for your work and this awesome library. I also started trying to build something with an Arduino Nano 33 BLE. I saw that this issue is still open. I just tested and compile my sketch with the CustomSettings branch that you posted in this message (https://github.com/lathoub/Arduino-BLE-MIDI/issues/55#issuecomment-1200351828).
Unfortunately, the following error appears:

"In file included from C:\Users\MidiBlePairingandButton\MidiBlePairingandButton.ino:5:0:
C:\Users\Arduino\libraries\BLE-MIDI\src/hardware/BLEMIDI_ArduinoBLE.h:140:1: error: expected class-name before '{' token
 {
 ^
C:\Users\Arduino\libraries\BLE-MIDI\src/hardware/BLEMIDI_ArduinoBLE.h:165:1: error: expected class-name before '{' token
 {
 ^
C:\Users\Arduino\libraries\BLE-MIDI\src/hardware/BLEMIDI_ArduinoBLE.h: In member function 'bool bleMidi::BLEMIDI_ArduinoBLE<_Settings>::begin(const char*, bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ArduinoBLE<_Settings>, _Settings>*)':
C:\Users\Arduino\libraries\BLE-MIDI\src/hardware/BLEMIDI_ArduinoBLE.h:204:9: error: 'class BLELocalDevice' has no member named 'setCallbacks'
     BLE.setCallbacks(new MyServerCallbacks<_Settings>(this));
         ^~~~~~~~~~~~
C:\Users\Arduino\libraries\BLE-MIDI\src/hardware/BLEMIDI_ArduinoBLE.h:206:15: error: 'class BLECharacteristic' has no member named 'setCallbacks'
     _midiChar.setCallbacks(new MyCharacteristicCallbacks<_Settings>(this));
               ^~~~~~~~~~~~
Multiple libraries were found for "BLEMIDI_Transport.h"
  Used: C:\Users\Arduino\libraries\BLE-MIDI
Compilation error: exit status 1"

Nevetheless, I can compile and send midi notes to my Windows machine using this branch (https://github.com/lathoub/Arduino-BLE-MIDI/tree/Arduino-Nano-BLE-33)

lathoub, 2025-02-02:
CustomSettings branch is still being worked on