lathoub/Arduino-BLE-MIDI #66: Password when pairing.

jhsa, 2022-09-20:
I couldn't find anything about requesting a password when pairing the BLE-MIDI with another device, so I guess it could be a legitimate feature request?
The reason for this  is that if we don't pair a device immediately when we are on stage, and as it is not protected by a password, anyone from the audience could pair to the ESP32 running the BLE-MIDI library, and together with a midi app on his/her phone, cause a big caos on stage. I think a password is very important to prevent something like that to happen on stage. MIDI is made for musicians, and musicians perform for a lot of people..
Thanks in advance for looking into it.

lathoub, 2022-10-22:
That would be a great addition - feel free to add password pairing to this lib and suggest a PR

jhsa, 2022-10-24:
> That would be a great addition - feel free to add password pairing to this lib and suggest a PR

It would be a great addition, for safety reasons, that is why I suggested it. :) 
But I don't know how to do it. If I did, I wouldn't need to suggest it, I would have already done it. ;) 
Unfortunately, I am still not at your level as far as programming is concerned.
Thanks for this excellent library by the way.

RobertoHE, 2022-11-29:
BLE client can accept a user-defined method for setting an authorized pairing by password, ping or whatever in the develop branch https://github.com/RobertoHE/Arduino-BLE-MIDI/tree/CustomSettings (this branch is under pull request yet).
You can add in your main code a configuration Struct (see the example of the branch) and add a custom callback function with your password or whatever you want and you can set the pairing capabilities of your BT device (screen, keyboard, etc).