lathoub/Arduino-BLE-MIDI #67: esp32 BLE-Midi and BluetoothSerial in the same project

oiresgleichen, 2022-10-03:
is it possible to use both librarys in one project?
if I use both libraries... It works for a while but after minutes it disconnects itself from the mac. and I can't reconnect without restart my esp. is there a way to solve this issue?
thanks

lathoub, 2022-10-03:
I don't use them together - if you can dig a big deeper why the disconnect, you might find the issue and fix it 🤞

RobertoHE, 2022-11-29:
It's possible. @oiresgleichen 
You need to include the 2 transport in your code. After that, create the 2 objects (serial and BLE) **WITHOUT** using the `BLEMIDI_CREATE_DEFAULT_INSTANCE()` macro, it is necessary to create the 2 objects using the explicit form of template constructor filling the `< >` with the corresponding hardware transport layer.