lathoub/Arduino-BLE-MIDI #70: BLE MIDI Client, ESP32-S3, keeps sending serial when not connected

jhsa, 2022-10-27:
I am trying to make an BLE MIDI to USB Midi and Serial midi adapter. Everything works, but when it disconnects, the ESP32-S3 starts sending something on the serial pin, until it connects again.
Is this a bug? Is it possible to disable this serial in the library?
Thanks

```
#include <Arduino.h>
#include <MIDI.h>
#include <Adafruit_TinyUSB.h>
#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_Client_ESP32.h>

BLEMIDI_CREATE_DEFAULT_INSTANCE(); //Connect to first server found
MIDI_CREATE_INSTANCE(HardwareSerial, Serial,  MidiSerial);

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MidiUsb);

int Led = 42; //modify for match with yout board

bool isConnected = false;

void setup()
{
 

  BLEMIDI.setHandleConnected([]()
  {
    Serial.println("---------CONNECTED---------");
    isConnected = true;
    digitalWrite(Led, HIGH);
  });

  BLEMIDI.setHandleDisconnected([]()
  {
    Serial.println("---------NOT CONNECTED---------");
    isConnected = false;
    digitalWrite(Led, LOW);
    
  });

  pinMode(Led, OUTPUT);
  digitalWrite(Led, LOW);

   TinyUSBDevice.setManufacturerDescriptor("My_Manufacturer");
  TinyUSBDevice.setProductDescriptor("BLE_USB_Aapter");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MidiUsb.begin(MIDI_CHANNEL_OMNI);

  MIDI.begin(MIDI_CHANNEL_OMNI);
  MidiSerial.begin(MIDI_CHANNEL_OMNI);

  MidiSerial.turnThruOff();
  MIDI.turnThruOff();
  MidiUsb.turnThruOff();
}

void loop()
{
  if (MidiSerial.read()) {

    MIDI.send(MidiSerial.getType(),
              MidiSerial.getData1(),
              MidiSerial.getData2(),
              MidiSerial.getChannel());

    MidiUsb.send(MidiSerial.getType(),
                 MidiSerial.getData1(),
                 MidiSerial.getData2(),
                 MidiSerial.getChannel());
  }
  if (MIDI.read()) {

   if(isConnected) {
    MidiSerial.send(MIDI.getType(),
                    MIDI.getData1(),
                    MIDI.getData2(),
                    MIDI.getChannel());
   }

    MidiUsb.send(MIDI.getType(),
                 MIDI.getData1(),
                 MIDI.getData2(),
                 MIDI.getChannel());
  }

  if (MidiUsb.read()) {
    MIDI.send(MidiUsb.getType(),
              MidiUsb.getData1(),
              MidiUsb.getData2(),
              MidiUsb.getChannel());

 if(isConnected) {
    MidiSerial.send(MidiUsb.getType(),
                    MidiUsb.getData1(),
                    MidiUsb.getData2(),
                    MidiUsb.getChannel());

  }
   }
}

```

jhsa, 2022-10-28:
Solved, commented all Serial Prints in "BLEMIDI_Client_ESP32.h"

closed 2022-10-28 by jhsa

lathoub, 2022-10-28:
> Solved, commented all Serial Prints in "BLEMIDI_Client_ESP32.h"

a classic error :-) (still happens to me once in a while)

jhsa, 2022-10-28:
> > Solved, commented all Serial Prints in "BLEMIDI_Client_ESP32.h"
> 
> a classic error :-) (still happens to me once in a while)

As a beginner, I am a little afraid of editing the libraries. Isn't that a way of moving all those serial lines to the examples themselves?
I think that all the mods I did will be erased next time I update the library, right?

lathoub, 2022-10-28:
your code sits aside from the library, so nothing will be reset next time you update.
Unless you made changes to the library code itself

jhsa, 2022-10-28:
> your code sits aside from the library, so nothing will be reset next time you update. Unless you made changes to the library code itself

Yes, that is what I meant. As I said above, commented all Serial Prints in "BLEMIDI_Client_ESP32.h", and of course also, no serial stuff on my code as well.