lathoub/Arduino-BLE-MIDI #72: BLE-MIDI client: Changes in NimBLE version 1.4.1

MicroMidi, 2022-11-07:
The NimBLE library changed its default response behaviour when (un-)subscribing to notifications/indications in Version 1.4.1 (https://github.com/h2zero/NimBLE-Arduino/pull/470). This leads to reconnection problems with a BLE-MIDI server I use in combination with the BLE-MIDI client code. But the response behaviour can be controlled by parameters in the NimBLE library (https://github.com/h2zero/NimBLE-Arduino/issues/473#issuecomment-1304812406).

Is it possible to introduce a user defined parameter in the BLE-MIDI client code to let the user decide whether a response should be generated on (un-)subscribing to notifications/indications or not?

Thanks in advance your support!

jhsa, 2022-11-25:
> The NimBLE library changed its default response behaviour when (un-)subscribing to notifications/indications in Version 1.4.1 ([h2zero/NimBLE-Arduino#470](https://github.com/h2zero/NimBLE-Arduino/pull/470)). This leads to reconnection problems with a BLE-MIDI server I use in combination with the BLE-MIDI client code. But the response behaviour can be controlled by parameters in the NimBLE library ([h2zero/NimBLE-Arduino#473 (comment)](https://github.com/h2zero/NimBLE-Arduino/issues/473#issuecomment-1304812406)).
> 
> Is it possible to introduce a user defined parameter in the BLE-MIDI client code to let the user decide whether a response should be generated on (un-)subscribing to notifications/indications or not?
> 
> Thanks in advance your support!

Not sure this question is in the right place, and if it is not, please forgive me.
Using this library, is it possible to boot an ESP32 as BLE MIDI client or Server, depending on the value of a variable at power ON?
The reason I want to do this is that I have built a device as client to connect to another ESP32 based device configured as server. But if I want to connect the client device to my tablet, it doesn't connect unless it would be configured as server instead of client. I really hope this is possible, otherwise I would have to upload the code every time I need to change its behavior. If this is possible, could you please point me in the right direction? 
Thanks.

MicroMidi, 2022-11-25:
> Not sure this question is in the right place, and if it is not, please forgive me.
As I understood this Github structure right this area is designated for issues - and each issue should have it's own thread...
You already asked this question in "Discussions"-area - and this is (from my point of view) the right place to discuss all questions regarding this library👍.

jhsa, 2022-11-25:
> As I understood this Github structure right this area is designated for issues - and each issue should have it's own thread... You already asked this question in "Discussions"-area - and this is [...]

I understand, the problem is that it seems no one is answering questions on the discussions area :( I asked a couple questions weeks ago and they are left with no answers. I also did a lot of googling which return zero results. I would like to complete my project but not able to.  I think there is another library that allows for this but the midi implementation is not as good as this one.
I do apologize again.. Please feel free to remove these posts.. I hope someone has some answers on the discussions area.

lathoub, 2025-02-02:
@MicroMidi your code is now in the main branch, and should work.

(changes to the NimBLE libs are now also incorporated, mostly on the security side)