lathoub/Arduino-BLE-MIDI #76: Problem: BLEMIDI_CREATE_INSTANCE with custom names

rigr, 2023-02-12:
Hello, first of all thank you for all the work you put in this project.
I have a problem with your library, specially when setting up an instance with a custom name.
the following code compiles on a ESP32 Dev Module but I'd be happy to be able to use "BLMisConnected " so the output part can decide to send BLE - MIDI or not. The problem is that I use AppleMIDI and DIN-MIDI as well, so I need to give the MIDI instances different names - DIN-MIDI is "MIDI1", AppleMIDI is "MIDI" and I'd like to call the BLE-MIDI "BLUETOOTHMIDI".
I put the compiler error as a comment next to the command I tried.

How can I sort that out?
Thanks in advance!

```
#include <BLEMIDI_Transport.h>

//#include <hardware/BLEMIDI_ESP32_NimBLE.h>
#include <hardware/BLEMIDI_ESP32.h>
//#include <hardware/BLEMIDI_nRF52.h>
//#include <hardware/BLEMIDI_ArduinoBLE.h>

//BLEMIDI_CREATE_DEFAULT_INSTANCE()
BLEMIDI_CREATE_INSTANCE("BLMIDIESP32", BLUETOOTHMIDI)

unsigned long t0 = millis();
bool BLMisConnected = true;  //set it to yes to see it running

// -----------------------------------------------------------------------------
// When BLE connected, LED will turn on (indication that connection was successful)
// When receiving a NoteOn, LED will go out, on NoteOff, light comes back on.
// This is an easy and conveniant way to show that the connection is alive and working. 
// -----------------------------------------------------------------------------
void setup()
{
  BLUETOOTHMIDI.begin();

  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);

  // uncommented by me
  //BLEMIDI.setHandleConnected([]() {        // 'BLEMIDI' was not declared in this scope
  //BLUETOOTHMIDI.setHandleConnected([]() {  // 'class midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32>, 
                                             //  bleMidi::MySettings>' has no member named 'setHandleConnected'
  //  BLMisConnected = true;
  //  digitalWrite(4, HIGH);
  // });

  // uncommented by me
  // BLEMIDI.setHandleDisconnected([]() {
  //  BLMisConnected = false;
  //  digitalWrite(4, LOW);
  // });

  BLUETOOTHMIDI.setHandleNoteOn([](byte channel, byte note, byte velocity) {
    digitalWrite(4, LOW);
  });
  BLUETOOTHMIDI.setHandleNoteOff([](byte channel, byte note, byte velocity) {
    digitalWrite(4, HIGH);
  });
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  BLUETOOTHMIDI.read();

  if (BLMisConnected && (millis() - t0) > 1000)
  {
    t0 = millis();

    BLUETOOTHMIDI.sendNoteOn (60, 100, 1); // note 60, velocity 100 on channel 1
  }
}

```

lathoub, 2023-02-12:
The `#include <hardware/BLEMIDI_ESP32.h>` indeed creates errors (ESP32 repo has recently changed, that may be the source of the error), but `NimBLE` works fine.

Change the code to:

```
#include <BLEMIDI_Transport.h>

#include <hardware/BLEMIDI_ESP32_NimBLE.h>
//#include <hardware/BLEMIDI_ESP32.h>
//#include <hardware/BLEMIDI_nRF52.h>
//#include <hardware/BLEMIDI_ArduinoBLE.h>

```

rigr, 2023-02-12:
Thanks for your fast reply, I feel honored. :)

I'd like to use the "BLMisConnected" variable so uncommented this:
```

BLEMIDI.setHandleConnected([]() {  // 'class midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32>, 
                                             //  bleMidi::MySettings>' has no member named 'setHandleConnected'
    BLMisConnected = true;
    digitalWrite(4, HIGH);
  });
```

It does not compile - 
28:17: error: 'class midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32_NimBLE>, bleMidi::MySettings>' has no member named 'setHandleConnected'
   BLUETOOTHMIDI.setHandleConnected([]() {  // 'class midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32>,
   
and when renamed to BLEMIDI I get:
   28:3: error: 'BLEMIDI' was not declared in this scope
   BLEMIDI.setHandleConnected([]() {  // 'class midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32>,

:(
How can I achieve to use that variable?
Thanks!

lathoub, 2023-02-12:
Look at the creation MACRO: 
https://github.com/lathoub/Arduino-BLE-MIDI/blob/929c2fc04962672ddba903a618601bea44ee1f3e/src/hardware/BLEMIDI_ESP32_NimBLE.h#L163

This prefixes the chosen name (`BLUETOOTHMIDI`) with `BLE` and becomes: `BLEBLUETOOTHMIDI`

Working sketch:

```cpp
#include <BLEMIDI_Transport.h>

#include <hardware/BLEMIDI_ESP32_NimBLE.h>
//#include <hardware/BLEMIDI_ESP32.h>
//#include <hardware/BLEMIDI_nRF52.h>
//#include <hardware/BLEMIDI_ArduinoBLE.h>

//BLEMIDI_CREATE_DEFAULT_INSTANCE()
BLEMIDI_CREATE_INSTANCE("BLMIDIESP32", BLUETOOTHMIDI)

unsigned long t0 = millis();
bool BLMisConnected = true;  //set it to yes to see it running

// -----------------------------------------------------------------------------
// When BLE connected, LED will turn on (indication that connection was successful)
// When receiving a NoteOn, LED will go out, on NoteOff, light comes back on.
// This is an easy and conveniant way to show that the connection is alive and working. 
// -----------------------------------------------------------------------------
void setup()
{
  BLUETOOTHMIDI.begin();

  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);

  // uncommented by me
  BLEBLUETOOTHMIDI.setHandleConnected([]() {   
    BLMisConnected = true;
    digitalWrite(4, HIGH);
   });

  // uncommented by me
  // BLEMIDI.setHandleDisconnected([]() {
  //  BLMisConnected = false;
  //  digitalWrite(4, LOW);
  // });

  BLUETOOTHMIDI.setHandleNoteOn([](byte channel, byte note, byte velocity) {
    digitalWrite(4, LOW);
  });
  BLUETOOTHMIDI.setHandleNoteOff([](byte channel, byte note, byte velocity) {
    digitalWrite(4, HIGH);
  });
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  BLUETOOTHMIDI.read();

  if (BLMisConnected && (millis() - t0) > 1000)
  {
    t0 = millis();

    BLUETOOTHMIDI.sendNoteOn (60, 100, 1); // note 60, velocity 100 on channel 1
  }
}
```

Good luck!

rigr, 2023-02-12:
Thank you very much. This worked perfectly!

closed 2023-02-14 by rigr