lathoub/Arduino-BLE-MIDI #82: Active Sensing

UncleRoel, 2023-11-17:
Active sensing (or other system realtime messages) is not implemented. 

Would be great to have it implemented so connected devices can show that the device is active and connected.

lathoub, 2023-11-17:
Very good comment, I have implemented ActiveSensing as part of the underlying library [FortySevenEffects Arduino MIDI Library](https://github.com/FortySevenEffects/arduino_midi_library) and a new version is soon to be released - stay tuned as I will rev this lib once the base library is updated.

lathoub, 2025-03-29:
> It's nice to se that you respond to something.

@jaca2300 I read your passive aggressive statement. Do not confuse a best effort open-source project with commercial SLA-based software.