lathoub/Arduino-BLE-MIDI #84: BLE Client Problem with Adafruit TinyUSB

ErfanDL, 2024-02-27:
Hi. I want set the midi output to adafruit tinyusb host but get compile error.
the code:

```
#include <Arduino.h>
#include <BLEMIDI_Transport.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>
#include <hardware/BLEMIDI_Client_ESP32.h>

//#include <hardware/BLEMIDI_ESP32_NimBLE.h>
//#include <hardware/BLEMIDI_ESP32.h>
//#include <hardware/BLEMIDI_nRF52.h>
//#include <hardware/BLEMIDI_ArduinoBLE.h>
Adafruit_USBD_MIDI usb_midi;

BLEMIDI_CREATE_DEFAULT_INSTANCE(); //Connect to first server found
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

//BLEMIDI_CREATE_INSTANCE("",MIDI)                  //Connect to the first server found
//BLEMIDI_CREATE_INSTANCE("f2:c1:d9:36:e7:6b",MIDI) //Connect to a specific BLE address server
//BLEMIDI_CREATE_INSTANCE("MyBLEserver",MIDI)       //Connect to a specific name server

#ifndef LED_BUILTIN
#define LED_BUILTIN 2 //modify for match with yout board
#endif

void ReadCB(void *parameter);       //Continuos Read function (See FreeRTOS multitasks)

unsigned long t0 = millis();
bool isConnected = false;

/**
 * -----------------------------------------------------------------------------
 * When BLE is connected, LED will turn on (indicating that connection was successful)
 * When receiving a NoteOn, LED will go out, on NoteOff, light comes back on.
 * This is an easy and conveniant way to show that the connection is alive and working.
 * -----------------------------------------------------------------------------
*/
void setup()
{
  Serial.begin(115200);
  MIDI.begin(MIDI_CHANNEL_OMNI);

  BLEMIDI.setHandleConnected([]()
                             {
                               Serial.println("---------CONNECTED---------");
                               isConnected = true;
                               digitalWrite(LED_BUILTIN, HIGH);
                             });

  BLEMIDI.setHandleDisconnected([]()
                                {
                                  Serial.println("---------NOT CONNECTED---------");
                                  isConnected = false;
                                  digitalWrite(LED_BUILTIN, LOW);
                                });

  MIDI.setHandleNoteOn([](byte channel, byte note, byte velocity)
                       {
                         Serial.print("NoteON: CH: ");
                         Serial.print(channel);
                         Serial.print(" | ");
                         Serial.print(note);
                         Serial.print(", ");
                         Serial.println(velocity);
                         digitalWrite(LED_BUILTIN, LOW);
                       });
  MIDI.setHandleNoteOff([](byte channel, byte note, byte velocity)
                        {
                        digitalWrite(LED_BUILTIN, HIGH);
                         });

  xTaskCreatePinnedToCore(ReadCB,           //See FreeRTOS for more multitask info  
                          "MIDI-READ",
                          3000,
                          NULL,
                          1,
                          NULL,
                          1); //Core0 or Core1

    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
    //MIDI.read();  // This function is called in the other task

  if (isConnected && (millis() - t0) > 1000)
  {
    t0 = millis();

    MIDI.sendNoteOn(60, 100, 1); // note 60, velocity 100 on channel 1
    vTaskDelay(250/portTICK_PERIOD_MS);
    MIDI.sendNoteOff(60, 0, 1);
  }
}

/**
 * This function is called by xTaskCreatePinnedToCore() to perform a multitask execution.
 * In this task, read() is called every millisecond (approx.).
 * read() function performs connection, reconnection and scan-BLE functions.
 * Call read() method repeatedly to perform a successfull connection with the server 
 * in case connection is lost.
*/
void ReadCB(void *parameter)
{
//  Serial.print("READ Task is started on core: ");
//  Serial.println(xPortGetCoreID());
  for (;;)
  {
    MIDI.read(); 
    vTaskDelay(1 / portTICK_PERIOD_MS); //Feed the watchdog of FreeRTOS.
    //Serial.println(uxTaskGetStackHighWaterMark(NULL)); //Only for debug. You can see the watermark of the free resources assigned by the xTaskCreatePinnedToCore() function.
  }
  vTaskDelay(1);
}
```

compile error:

```
In file included from c:\Users\STRIX-PC\Documents\Arduino\libraries\MIDI_Library\src/MIDI.h:35,
                 from c:\Users\STRIX-PC\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/BLEMIDI_Transport.h:7,
                 from C:\Users\STRIX-PC\AppData\Local\Temp\.arduinoIDE-unsaved2024127-9620-yia50d.15mhn\sketch_feb27a\sketch_feb27a.ino:29:
c:\Users\STRIX-PC\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:101:73: error: conflicting declaration 'midi::MidiInterface<midi::SerialMIDI<Adafruit_USBD_MIDI> > MIDI'
     MIDI_NAMESPACE::MidiInterface<MIDI_NAMESPACE::SerialMIDI<Type>> Name((MIDI_NAMESPACE::SerialMIDI<Type>&)serial##Name);
                                                                         ^
C:\Users\STRIX-PC\AppData\Local\Temp\.arduinoIDE-unsaved2024127-9620-yia50d.15mhn\sketch_feb27a\sketch_feb27a.ino:41:1: note: in expansion of macro 'MIDI_CREATE_INSTANCE'
 MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
 ^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\STRIX-PC\AppData\Local\Temp\.arduinoIDE-unsaved2024127-9620-yia50d.15mhn\sketch_feb27a\sketch_feb27a.ino:33:
c:\Users\STRIX-PC\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/hardware/BLEMIDI_Client_ESP32.h:636:33: note: previous declaration as 'midi::MidiInterface<bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_Client_ESP32>, bleMidi::MySettings> MIDI'
     BLEMIDI_CREATE_INSTANCE("", MIDI)
                                 ^~~~
c:\Users\STRIX-PC\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/hardware/BLEMIDI_Client_ESP32.h:630:145: note: in definition of macro 'BLEMIDI_CREATE_INSTANCE'
     MIDI_NAMESPACE::MidiInterface<BLEMIDI_NAMESPACE::BLEMIDI_Transport<BLEMIDI_NAMESPACE::BLEMIDI_Client_ESP32>, BLEMIDI_NAMESPACE::MySettings> Name((BLEMIDI_NAMESPACE::BLEMIDI_Transport<BLEMIDI_NAMESPACE::BLEMIDI_Client_ESP32> &)BLE##Name);
                                                                                                                                                 ^~~~
C:\Users\STRIX-PC\AppData\Local\Temp\.arduinoIDE-unsaved2024127-9620-yia50d.15mhn\sketch_feb27a\sketch_feb27a.ino:40:1: note: in expansion of macro 'BLEMIDI_CREATE_DEFAULT_INSTANCE'
 BLEMIDI_CREATE_DEFAULT_INSTANCE(); //Connect to first server found
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

lathoub, 2024-02-27:
This library was developed prior to Adafruit_TinyUSB and only does MIDI - the Adafruit lib does much more. Why do you want to mix them?

ErfanDL, 2024-02-28:
> This library was developed prior to Adafruit_TinyUSB and only does MIDI - the Adafruit lib does much more. Why do you want to mix them?

thanks for reply. I want to get BLE midi from a device and send it to ESP32 usb host for android.

closed 2025-02-02 by lathoub