lathoub/Arduino-BLE-MIDI #86: Trying to get bluetooth midi for Arduino Nano 33 BLE sense rev2

nicolasvair, 2024-03-20:
Hi,

I'm trying to use this lib with a Arduino Nano 33 BLE sense rev2.

I'm quite new to this. I used the include #include <hardware/BLEMIDI_nRF52.h> because that's what this board is using. I compiled ok but impossible to connect from my mac.  I can't see the board as bluetooth midi. 

After searching for a bit I ended up looking at the code for "[hardware/BLEMIDI_nRF52.h](https://github.com/lathoub/Arduino-BLE-MIDI/blob/929c2fc04962672ddba903a618601bea44ee1f3e/src/hardware/BLEMIDI_nRF52.h)" and see that it is written as "in development" and that all the "advertising" lines are commented out. I wonder if i should uncomment them and use the "#include <bluefruit.h>" ??

Or maybe just return the board and order a esp32 that seems way more simple to use. I chose this one because I liked the sensors and AI capabilities for future but my main usage would be midi over bluetooth.

Any help ?
Thank you !

lathoub, 2024-03-20:
try this branch: https://github.com/lathoub/Arduino-BLE-MIDI/tree/Arduino-Nano-BLE-33 (and let me know if it works)

nicolasvair, 2024-03-20:
Hi, 

Thank you very much for your fast answer. Yes this branch works fine. 

With the main one, compilation wouldn't work with "#include <hardware/BLEMIDI_ArduinoBLE.h>"
So I thought I should use "#include <hardware/BLEMIDI_nRF52.h>" because my board use a nRF52840 (and it was the only one compiling)

But with the branch you sent me, I was able to compile with "#include <hardware/BLEMIDI_ArduinoBLE.h>" and now it is seen by my computer. Perfect ! 

For a newbie like I am, this phrase wasn't clear : 
`When using ESP32 consider using NimBLE (NimBLE-Arduino). When using the Arduino NANO 33 BLE or Arduino NANO RP2040 Connect, you must install ArduinoBLE`

As I'm using a Arduino NANO 33 BLE, I understood that I had to install ArduinoBLE (which I did), but I didn't understand that I had to compile with "#include <hardware/BLEMIDI_ArduinoBLE.h>". Maybe it could be made more clear. 

Any way, thank you very much for your help. 

I'll go further in my project now. Let me know if I can help !

closed 2024-03-24 by lathoub