lathoub/Arduino-BLE-MIDI #87: Issue with Ableton and midi "feedback"

nicolasvair, 2024-04-14:
Hi, 

First thank you for this fantastic repo. I'm doing a "4 knobs" wireless controller using encoders and led ring. 

[image]

The idea is to be able to control a parameter from various devices and have led feedback on each of them. The other being [midi fighter twister](https://store.djtechtools.com/products/midi-fighter-twister) is also based on encoders and led rings.

I came across a strange behavior, if arduino is connected to computer through BLE-midi, I have a strange behavior on ableton parameters mapped to midi CC. Not sure how to describe it, it is jerky. 

https://github.com/lathoub/Arduino-BLE-MIDI/assets/79313360/8250912d-22b1-465a-9805-a4340fd8c08a

Important to note, this is independent of my code. I can upload a bare minimum code for connecting, and still have the issue. Even without : 
` MIDI.setHandleControlChange(OnControlChange);
`

For example I have the issue with this code : 

`#include <BLEMIDI_Transport.h>

//#include <hardware/BLEMIDI_ESP32_NimBLE.h>
//#include <hardware/BLEMIDI_ESP32.h>
//#include <hardware/BLEMIDI_nRF52.h>
#include <hardware/BLEMIDI_ArduinoBLE.h>
#include "Arduino_BMI270_BMM150.h"

BLEMIDI_CREATE_DEFAULT_INSTANCE()

unsigned long t0 = millis();
bool isConnected = false;

void setup()
{
  // MIDI INIT
  MIDI.begin();

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  BLEMIDI.setHandleConnected([]() {
    isConnected = true;
    digitalWrite(LED_BUILTIN, HIGH);
    MIDI.sendControlChange(1, 1, 1);
  });

  BLEMIDI.setHandleDisconnected([]() {
    isConnected = false;
    digitalWrite(LED_BUILTIN, LOW);
  });

}
void loop()
{  
  MIDI.read();
}
`

Maybe related to how the arduino send a "confirmation" of the received CC from ableton ?
When I look into details, ableton sends values like : 24, 25, 27, **22**, 29, 32, 35 ...

But it only happens with this device and this repo. I thought it may be related to bluetooth latency, but I don't have any issue using TouchOsc on iPhone...

Any hints where to look at ?

Thank you 🙏

lathoub, 2024-04-14:
Hi, nice work!
This issue might relate to the [underlying MIDI library](https://github.com/FortySevenEffects/arduino_midi_library). Have you searched the issues section for CC related issues?

(This library is a thin wrapper around FortySevenEffects MIDI lib)

nicolasvair, 2024-04-14:
Hi, thank you for the quick reply !

I just searched the issues but could not find anything related.

lathoub, 2024-04-17:
I'm going to recommend to create a similar issue (and also point to this one) in the underlying [library git](https://github.com/FortySevenEffects/arduino_midi_library). The community there is much better to respond to CC problems.

nicolasvair, 2024-04-17:
Ok, thank you then

If anyone is looking for a solution, the new issue is here : https://github.com/FortySevenEffects/arduino_midi_library/issues/346

nicolasvair, 2024-05-08:
Hi, 

So after some more research, it seems that the midi gets "reflected" like a midi thru situation.
(like [this post](https://github.com/FortySevenEffects/arduino_midi_library/issues/147))

Unfortunately, the function   MIDI.turnThruOff(); didn't help :(

[image]

lathoub, 2024-05-09:
Bizar, MIDI thru has been switched pff

https://github.com/lathoub/Arduino-BLE-MIDI/blob/929c2fc04962672ddba903a618601bea44ee1f3e/src/BLEMIDI_Transport.h#L55

Can you check in the [underlying lib](https://github.com/FortySevenEffects/arduino_midi_library) if MIDI Thru has been switched off (breakpoint or log message)