lathoub/Arduino-BLE-MIDI #88: Can't compile anymore on arduino 2.3 - conversion from 'String' to non-scalar type 'std::string'

antonvh, 2024-06-05:
My code used to work, but sadly, I hit the update button. :/

Now compiling my sketch with BLE-MIDI returns this:
```
In file included from /Users/anton/Documents/Arduino/blemidi2/blemidi2.ino:2:
/Users/anton/Documents/Arduino/libraries/Arduino-BLE-MIDI/src/hardware/BLEMIDI_ESP32.h: In member function 'virtual void bleMidi::MyCharacteristicCallbacks::onWrite(BLECharacteristic*)':
/Users/anton/Documents/Arduino/libraries/Arduino-BLE-MIDI/src/hardware/BLEMIDI_ESP32.h:116:55: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
  116 |         std::string rxValue = characteristic->getValue();
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: exit status 1
```

antonvh, 2024-06-05:
Solved. Found out how to use nimBLE with
#include <hardware/BLEMIDI_ESP32_NimBLE.h>

closed 2024-06-05 by antonvh

wiemax1093, 2024-08-22:
imo this is still a bug and needs to be fixed (just first experienced this under IDE 2.3.0/ fresh isntall and update of everything),

also for a workaround (for everyone), change line 116 in BLEMIDI_ESP32.h from:
`std::string rxValue = characteristic->getValue();`
to
`String rxValue = characteristic->getValue();`

thx to a similar problem over here that explained the root cause https://forum.arduino.cc/t/compilation-error-on-code-that-was-compiling-before-conversion-from-string-to-non-scalar-type/1265323 // https://github.com/espressif/arduino-esp32/pull/8724

@lathoub maybe you can implement the change directly in a future update

lathoub, 2024-08-23:
@wiemax1093 thx for signaling. If you make a PR, I'm happy to accept