lathoub/Arduino-BLE-MIDI #96: BLE-MIDI/src/hardware/BLEMIDI_ESP32_NimBLE.h compile error

RCPilot1604, 2025-01-02:
Attempted to compile a previously working piece of code which now produces the following compilation error. 
The error is in line 144 and 145 of [Arduino-BLE-MIDI/src/hardware/BLEMIDI_ESP32_NimBLE.h](https://github.com/lathoub/Arduino-BLE-MIDI/blob/master/src/hardware/BLEMIDI_ESP32_NimBLE.h). 

I suspect that there was a migration of the [h2zero/NimBLE-Arduino](https://registry.platformio.org/libraries/h2zero/NimBLE-Arduino) library, causing the code to break. 
```
In file included from src/main.cpp:4:
.pio/libdeps/esp32doit-devkit-v1/BLE-MIDI/src/hardware/BLEMIDI_ESP32_NimBLE.h: In member function 'bool bleMidi::BLEMIDI_ESP32_NimBLE::begin(const char*, bleMidi::BLEMIDI_Transport<bleMidi::BLEMIDI_ESP32_NimBLE>*)':
.pio/libdeps/esp32doit-devkit-v1/BLE-MIDI/src/hardware/BLEMIDI_ESP32_NimBLE.h:144:26: error: expected type-specifier before 'NimBLESecurity'
  144 |     auto _security = new NimBLESecurity();
      |                          ^~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/BLE-MIDI/src/hardware/BLEMIDI_ESP32_NimBLE.h:145:38: error: 'ESP_LE_AUTH_BOND' was not declared in this scope
  145 |     _security->setAuthenticationMode(ESP_LE_AUTH_BOND);
      |                                      ^~~~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
```

spDuchamp, 2025-01-07:
I can confirm I'm having the exact same problem, and when I roll back to version 2.1.0, the project compiles just fine.

MicroMidi, 2025-01-28:
May I asked if there will be a new official release of the brilliant library that leverages the features of the new major NimBLE 2.x.x version? Thanks in advance🙏

lathoub, 2025-01-28:
Hi @MicroMidi et al - good question. An official release of this library is difficult, as it relies on calls in the underlying [MIDI library](https://github.com/FortySevenEffects/arduino_midi_library) that have not been rolled up in a release. It has been in the air for a while, but we (@FortySevenEffects (*) and myself) have to organize a 'release party' to align all the lreleases of the libraries again (the base libraries and all the derived protocols). Maybe this is a good trigger for a concerted effort.

MicroMidi, 2025-01-30:
Hi @lathoub,
thank you for your reply. I wasn't aware of all the dependencies that need to be considered in order to have the furious “release party”. It would be really great to have an updated state of the art BLE-MIDI library, as the hard work you and the other library owners have put into developing these libraries will be carried into the future. Right now, users have to struggle with different library versions to find a solution that works for their project. And I'm pretty sure your user base is quite large.

Thanks again for your hard work and for your great contribution to the Arduino ESP32 community

Bernd

lathoub, 2025-02-02:
Problem fixed in master branch.
but... I do not have hardware to test for the moment, so please test for me!

the main security settings are
https://github.com/lathoub/Arduino-BLE-MIDI/blob/2e4b39fe49691e2c89ee69912e5fc117f8f1d916/src/hardware/BLEMIDI_ESP32_NimBLE.h#L142

parameters described [here](https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_device.html#aa8e340c02418771ce72dec758d560938): 

**Please let me know if this fix works**

JasonBugeja, 2025-02-07:
Hi @lathoub,
I have tried the fixed version.  Still gives the exact same error log. 

Thanks for your hard work and great contributions to the Arduino esp32 community

lathoub, 2025-02-08:
@JasonBugeja just checking: all is A-OK? (I just checked with a fresh checkout from github and all compiled correctly)

MicroMidi, 2025-02-09:
Hi @lathoub,

i tried to compile the BLE MIDI client demo code from @RobertoHE ("MidiBle_Client.ino") and get the following error while compiling:

_In file included from C:\Users\bernd\AppData\Local\Temp\arduino_modified_sketch_379318\MidiBle_Client.ino:31:
C:\Users\bernd\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/hardware/BLEMIDI_Client_ESP32.h:142:1: error: expected class-name before '{' token
  142 | {
      | ^
C:\Users\bernd\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/hardware/BLEMIDI_Client_ESP32.h: In member function 'void bleMidi::AdvertisedDeviceCallbacks::onResult(NimBLEAdvertisedDevice*)':
C:\Users\bernd\Documents\Arduino\libraries\Arduino-BLE-MIDI-master\src/hardware/BLEMIDI_Client_ESP32.h:168:119: error: no match for 'operator==' (operand types are 'const NimBLEAddress' and 'std::string' {aka 'std::__cxx11::basic_string<char>'})
  168 |         if (!(!specificTarget || (advertisedDevice->getName() == nameTarget.c_str() || advertisedDevice->getAddress() == nameTarget)))
      |                                                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^~ ~~~~~~~~~~
      |                                                                                                                    |     |
      |                                                                                                                    |     std::string {aka std::__cxx11::basic_string<char>}
      |                                                                                                                    const NimBLEAddress
In file included from C:\Users\bernd\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEClient.h:30,_

I tried that with the Espressif ESP32 v2.x and v3.x board framework and NimBLE library v2.2.1. Do you have any ideas how to solve that?
Thanks in advance for any hint

Bernd

lathoub, 2025-02-10:
The NimBLE library has changed its API, and I am unfamiliar with the BLE client usage.
I'm going to rely on RobertoHE to fix the code. 

@RobertoHE can you check the code?

Mainly here:
https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L168

https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L458

mix up on template parameters:

https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L371

RobertoHE, 2025-02-10:
Hello everyone.
Now I haven't any resources to test the solution with software or hardware until March. I can only give you some ideas or tips that you'll need to investigate.

1) return type of getAddress

https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L168

This line tries to check if the address or name matches the name that you choose, client will try to connect with that server. The error of @MicroMidi says that type mismatch. Would you verify that [NimbleBLEAddress->getAddres()](https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_advertised_device.html#acb3c376962b317b91628fb96ad215923) type return type matches with `string`? I think that the method has a casting method to string type ([toString()](https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_address.html#ac53b0cc945e1505d3e72d48164ad01d6)).

2) Possible error when CustomSetting merge.

I just compare my local repo with this master report
My Repo:

https://github.com/RobertoHE/Arduino-BLE-MIDI/blob/38fea5dfe32a93ce09cd271df03bfe8fb2407c9e/src/hardware/BLEMIDI_Client_ESP32.h#L580-L594 

Master Repo:

https://github.com/lathoub/Arduino-BLE-MIDI/blob/2e4b39fe49691e2c89ee69912e5fc117f8f1d916/src/hardware/BLEMIDI_Client_ESP32.h#L631-L643

At any moment you remove _setting Template class, @lathoub. If setting class is not defined in the constructor at creation time, some error may occur because some variables are not defined. 

The main change to work with customizable settings is based on the heritage of setting to midi->blemide->transport layer.

https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L30

In master repo, the settings are not used with the constructors. 
This kind of logic must be implemented at similar way and adapt to the specific setting or parameter of each layer. If you copy the logic of my local repo to all transport layers (heritage class for settings), you will save some problems in the future.

Please, check that settings are correctly used in the constructor in all layers, not only in Client transport layer.

Sorry for not being more useful until March.

RobertoHE, 2025-04-02:
Please. Check #102 pull request.