lathoub/Arduino-BLE-MIDI #99: Optional Serial Printing

samspencer5991, 2025-02-12:
Whilst some of it can be helpful, the large amount of debug/info printing by the client mode in particular can cause issues, especially when using the primary serial port for other uses. It would be more helpful if these Serial.print lines could be wrapped in perhaps a pre-processor conditional statement so that a global macro can be used to enable or disable them for debug and diagnostic purposes.

A good example is the client mode scanning results, where this could be wrapped in something like:

#ifdef BLE_SERIAL_DEBUG
.....
#endif

lathoub, 2025-02-12:
Are you referring to the `serial.print` in the examples or in the code?
(A quick scan indicates that there is only 1 serial.print in the code (which should be removed))

samspencer5991, 2025-02-12:
The file I'm currently looking at is "BLEMIDI_Client_ESP32.h" where client scanning results in constant serial printing. Specifically lines here:

192: Serial.print("Advertised Device found: ");
193: Serial.println(advertisedDevice->toString().c_str());

504: Serial.println("Scanning...");

This is just an example as I haven't done a full dive to inspect the rest of the library files.

lathoub, 2025-02-13:
What is the source of your code? A Release version (eg from the Arduino IDE Lib manager or PlatformIO) or the latest code from GitHub?

Here is the latest code in the area you mentioned (no serial prints):

https://github.com/lathoub/Arduino-BLE-MIDI/blob/4c2f4965fb29009653889287caf3d1546c130023/src/hardware/BLEMIDI_Client_ESP32.h#L189-L200

samspencer5991, 2025-03-21:
I'm using the PIO library manager, with the library properties showing v2.2

samspencer5991, 2025-04-02:
Closed due to using an older codebase

closed 2025-04-02 by samspencer5991