lathoub/Arduino-BLE-MIDI #101: Compile error on PlatformIO BLEMIDI_Namespace.h

dannyrubin, 2025-03-13:
In PlatformIO I've included the lastest library https://github.com/lathoub/Arduino-BLE-MIDI.git in my lib_deps.   Upon compiling I get several errors, the first in BLEMIDI_Namespace.h stating "expected declaration before '}' token [Ln 5 Col 44] "  and then several subsequent errors in BLEMIDI_Transport.h because of this , e.g. 'MIDI_NAMESPACE' is not a namespace-name [Ln 15 Col 17].

Can anyone suggest why this might be and how I can resolve it?
Thanks in advance

samspencer5991, 2025-03-21:
I've also experienced this when switching from the PIO v2.2 to directly GH sourced.

lathoub, 2025-03-21:
With the awaited release of the underlying MIDI library, PIO gets really confused. Best is to code the code from the Repo's at this stage (and not used the release libs)

samspencer5991, 2025-03-22:
I was previously using the PIO managed library version (although out dated), however, this issue is happening when I directly reference this GH repo using the https link due to needing to update the codebase.

samspencer5991, 2025-03-28:
@lathoub any update on this? Perhaps some more info about what issue with the forty seven effects library could be useful. Let me know if you need a hand with any of the upcoming library integration testing.

lathoub, 2025-03-29:
I have no plan to support PlatformIO, but willing to plan to do a release here so it works with 47effects base lib (v5.0.2) and wiht the latest versions of ESP32 NimBLE, ESP32 and ArduinoBLE)

The namespace error i'm unable to reproduce.

What hardware do you use? (which include do you use?)

```cpp
#include <hardware/BLEMIDI_ESP32_NimBLE.h>
#include <hardware/BLEMIDI_ESP32.h>
#include <hardware/BLEMIDI_ArduinoBLE.h>
```

lathoub, 2025-03-29:
I fixed the ArduinoBLE implementation (albeit still have to implement receiving from BLE).
Try with code from the branch main with the library Release 5.0.2 of the MIDI library and let me know if this works. 

If that works i'll create a 2.3 version

samspencer5991, 2025-03-30:
Ok will do and report back

samspencer5991, 2025-03-31:
Yep, still having compilation issues. Using the ESP32 platform by the way. I was initially using an older version of the library until I ran into various debug Serial.print actions which had been left in. These compilation errors came after updating this BLE-MIDI library.

It seems to be an incompatibility with the MIDI and NimBLE-Arduino libraries:

[image]

lathoub, 2025-04-01:
I've tested with a clean install using the Arduino IDE, using the latest Arduino ESP32 3.1.3, NimBLE 2.2.3 and MIDI 5.0.2. 

I do notice you used the BLE_Client example (correct?), which has issues for the moment (@RobertoHE will fix when he has time). Or @samspencer5991 maybe you can look at the template code and suggest a fix - happy to accept PR's

samspencer5991, 2025-04-02:
Yep, not using the client example, but using parts of the client library for configuration based switching. @RobertoHE I'm happy to take a look at the issues and suggest a fix, but I don't want to make things too messy if you're already working on something. I'll report back.

RobertoHE, 2025-04-03:
Check #102

mcelb1200, 2025-11-28:
> I have no plan to support PlatformIO, but willing to plan to do a release here so it works with 47effects base lib (v5.0.2) and wiht the latest versions of ESP32 NimBLE, ESP32 and ArduinoBLE)
> 
> The namespace error i'm unable to reproduce.
> 
> What hardware do you use? (which include do you use?)
> 
> #include <hardware/BLEMIDI_ESP32_NimBLE.h>
> #include <hardware/BLEMIDI_ESP32.h>
> #include <hardware/BLEMIDI_ArduinoBLE.h>

I think I might have a fix for the namespace error, as I've been struggling with it for a while. Try this??
https://github.com/lathoub/Arduino-BLE-MIDI/pull/105#issue-3664735665